scottrini/OctoPrint-PrusaLevelingGuide #1: Allow 7x7 leveling and configuration of probe retry count

Kelly12612, 2019-12-16:
Love the plugin, I see that you're issuing "G80 N9", the generic reprap doc I found said that only the values 3 and 7 are valid.  Either way, I'd like to be able to choose 7x7 instead of 3x3.

Related would be to add the configuration of the probe count between 3 and 10.
Thanks!

scottrini, 2019-12-16:
You're right, that's supposed to be N3.  But the gcode is already configurable.  Just go to settings and find the tab for Prusa Leveling Guide and change the gcode to whatever you want.  If you leave it just G80, it'll use whatever the firmware is configured for.

For the nylock mod, I always have it do 3x3 since you only have 8 screws to actually adjust, so even though it's checking the additional points, you can't really do anything about them.

scottrini, 2019-12-16:
I went ahead and fixed the default gcode to be N3, since that's correct, but there's nothing to add since you can already modify the gcode to perform whatever you'd like.

closed 2019-12-16 by scottrini