scottrini/OctoPrint-PrusaLevelingGuide #2: Feature request: allow adjustment of Z-Axis leadscrews

Spacemarine2018, 2019-12-18:
Even a perfectly leveled bed can appear to be heavily skewed (all 3 left or right points lower or higher than all 3 on the other side) if a user performs a Z-axis calibration AFTER leveling the bed in a MK3/S, MK2, Mk1. The reason ist, that the Z-endstops at the top of the printer are soft and stop the leadnut at slightly different positions each time. This brings the two Z-Axis leadscrews out of sync.

Some users are confused by this as they don't understand the mechanics behind this. It would therefore be desireable, if this plugin would suggest to rotate one of the Z-leadscrews by a certain amount of "clicks" to bring them back into sync. The amount of clicks depends on the angle of the Z-screw.

Since Prusa uses steppers with 1.8 ° step-angle (200 steps per rev.) and a stepper always skips 4 full steps at at time if overloaded or forcefully moved by hand, this results in 50 "clicks" per Z-axis revolution.

scottrini, 2019-12-19:
I'm not sure this is something I'd add.  None of my MK3S's are using the stock frame (bear frame), so I went ahead and tested this and after Z calibration, I didn't get different values like you're describing.

If you can come up with a specific mathematic way to determine the leadscrews are actually what need to be adjusted a formula for calculating how much, I'd consider adding it to the UI.  Or you could take a stab at a PR yourself.

Spacemarine2018, 2019-12-28:
If you have a very stiff mechanical stops for Z like the ones in the bear mod, I wouldn't expect that problem to arise. But can you reproduce the skewed plate, if you rotate one of the rods by one click by hand?

To calculate how much one Z-axis needs to be rotated, I would go like this:

skew = average(all left points)-average(all right points)
number of clicks to rotate = skew * scaling factor

The scaling factor depends on the thread pitch, the number of full steps per revolution of the steppers, the size of the bed, etc. Since this is the same for every I3 MK3/S printer, I wouldn't bother calculating it. I'd just rotate the one Z-Axis by one or two clicks, watch the result and adjust the scaling factor accordingly.

scottrini, 2020-01-05:
Sorry, I've thought about this and it's not something I'd work on.  It seems a little too open to interpretation and hard to get exact.  I even question exactly how to display it to the user.  Without this, the user can still remove the variance to a degree they're happy with.  An upcoming feature of this is to set a target variance in the settings and there be some sort of alert when the variance is above the target, so they can redo the leveling.  So if for some reason they redid the Z calibration and it went above the target, they'd be alerted and could redo it.

The most I'd be willing to do now is have a note somewhere in the README explaining this is a possibility.  If you'd like to draft that note, I'd take a look at it and consider including it.

closed 2020-01-05 by scottrini

Spacemarine2018, 2020-01-06:
Ok, makes sense. The note could probably include:

If you re-calibrate your Z-axis after leveling your bed, it might look like the whole left or right side of the bed is suddenly higer/lower than the other side. The reason for this is, that the Z-axis leadscrews might have changed their angular position relative to each other due to soft mechanical upper stops. If this is the case, try to rotate one of the leadscrews by hand by one or two clicks instead of re-adjusting the bed again.

scottrini, 2020-01-24:
Your note is going into the README.md for 1.0.6