scottrini/OctoPrint-PrusaLevelingGuide #4: [Feature request] More customization on gcode/profile

ShagoY, 2020-01-05:
I have added a gcode to wait for my PINDA probe to reach 35°c before leveling.

```
; Heat PINDA
G0 X100 Y100 Z0.15 F2000; PINDA warms faster near bed 
M860 S35; wait for PINDA temp. For most stable PINDA temp during mesh, set temp here 1-2 deg below your actual PINDA temp at end of mesh cal.
```

I would like to be able to move the nozzle before/during the heating of the bed/nozzle to speed up the process.

Could we have more customization here? Like adding a profile?

Thanks in advance.

scottrini, 2020-01-05:
I'm not sure this is something I'd include in the plugin.  Not sure how a profile would fit in.  The profiles you select for preheating are the temperature profiles built into octoprint.  You can't store additional data in these profiles, so they're out of the question.

There are already a few ways you can achieve this without any modifications.  You want the PINDA in that position before starting?  Send that gcode before starting an adjustment.

If you're worried about the PINDA not being completely preheated for a round of adjustments, just don't make any adjustments during the first round and give it some extra time to heat, then start making adjustments during the second round.

This plugin only performs the preheating at the very beginning.  It sets the temperature, then waits for the bed and tool to reach temperature, then begins.  It doesn't check preheating between every round of adjustments.

ShagoY, 2020-01-06:
I didn't realize you were using Octoprint's built-in profiles... 

Indeed, I just wanted it to approach the nozzle of the bed during the warm-up so it is not managed by your plugin.

Thanks for your answer :)

closed 2020-01-06 by ShagoY