scottrini/OctoPrint-PrusaLevelingGuide #5: no update of data after mesh bed leveling sequence

HarryT36, 2020-01-06:
there is no direct update of the shown data after a sequence. Also the update button does not work.

New data is shown after again starting a measuring sequence

scottrini, 2020-01-06:
Can you send a screenshot of what you have configured for the mesh leveling gcode?

HarryT36, 2020-01-06:
it is the standard config. 
[image]

Forgot to mention the version. This is 1.0.5
By the way: I'm running Octoprint on a raspi zero

HarryT36, 2020-01-06:
accidently closed => reopened

scottrini, 2020-01-06:
I only asked for that because the only way I can reproduce this (and it will cause the issue) is if the M400 is left out.  Can you watch the terminal and ensure it's sending the M400?

If it is, then it could be a hardcoded timeout I have set to make sure it's not watching gcode for too long.  I could potentially send you a file to try out.  But first make sure the M400 is being used.

HarryT36, 2020-01-06:
Here is the terminal content for the first run after starting up octoprint. (M400 is present)
Now the data was updated before the extruder starts to move.

Send: M105
Recv: ok T:34.7 /0.0 B:35.4 /0.0 T0:34.7 /0.0 @:0 B@:0 P:33.3 A:38.8
Send: M105
Recv: ok T:34.7 /0.0 B:35.4 /0.0 T0:34.7 /0.0 @:0 B@:0 P:33.3 A:38.7
Send: M105
Recv: ok T:34.7 /0.0 B:35.4 /0.0 T0:34.7 /0.0 @:0 B@:0 P:33.3 A:38.7
Send: G28 W
Send: M400
Recv: tmc2130_home_enter(axes_mask=0x01)
Recv:   0 step=62 mscnt= 994
Recv: tmc2130_goto_step 0 61 2 1000
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x01
Recv: tmc2130_home_enter(axes_mask=0x02)
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:   0 step= 1 mscnt=  31
Recv: tmc2130_goto_step 1 1 2 1000
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x02
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: ok
Send: G80 N3
Recv: ok
Send: G81
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: ok
Send: M113 S2
Recv: Num X,Y: 7,7
Recv: Z search height: 5.00
Recv: Measured points:
Recv:   0.17950  0.15189  0.13289  0.12250  0.12072  0.12756  0.14300
Recv:   0.14944  0.12702  0.11190  0.10406  0.10351  0.11025  0.12428
Recv:   0.13011  0.11102  0.09867  0.09306  0.09417  0.10202  0.11661
Recv:   0.12150  0.10389  0.09322  0.08950  0.09272  0.10289  0.12000
Recv:   0.12361  0.10562  0.09554  0.09339  0.09915  0.11284  0.13444
Recv:   0.13644  0.11621  0.10564  0.10472  0.11347  0.13188  0.15994
Recv:   0.16000  0.13567  0.12350  0.12350  0.13567  0.16000  0.19650
Recv: ok
Send: M105
Recv: ok
Recv: ok T:34.6 /0.0 B:35.2 /0.0 T0:34.6 /0.0 @:0 B@:0 P:33.2 A:38.7
Send: M105
Recv: ok T:34.6 /0.0 B:35.1 /0.0 T0:34.6 /0.0 @:0 B@:0 P:33.2 A:38.8
Send: G1 Z60 Y210 F6000
Recv: ok
Send: M105
Recv: ok T:34.5 /0.0 B:35.2 /0.0 T0:34.5 /0.0 @:0 B@:0 P:33.2 A:38.7
Send: M105
Recv: ok T:34.5 /0.0 B:35.0 /0.0 T0:34.5 /0.0 @:0 B@:0 P:33.2 A:38.7
Send: M105
Recv: ok T:34.5 /0.0 B:35.1 /0.0 T0:34.5 /0.0 @:0 B@:0 P:33.2 A:38.7

HarryT36, 2020-01-06:
Also repeating it with "COntinue" always lead to  an update of the dta.

It seems to be a problem on my side after some resets I have to perform due to missing steal sheet.

So there is NO bug.

Sorry for the inconvinience.

Bye Harry

closed 2020-01-06 by HarryT36