scottrini/OctoPrint-PrusaLevelingGuide #6: Bad bed page rendering

ochm, 2020-01-09:
I found bad rendering while displaying values on bed.
[image]

Octoprint 1.3.12,
Prusa Leveling Guide (1.0.5)

Trying in Chrome and Firefox ESR, both in latest versions.
With or without Octoprint plugin Themeify, the rendering is still bad.

Thank you for repair.

scottrini, 2020-01-09:
Sorry, this is only reproducible with Themeify.  We only test with the latest octoprint and Prusa Leveling Guide.

We have no plans to support every Themeify theme.  But the "Discorded" theme is known to work.

If you are still seeing issues with rendering without Themeify, which that issue looks like one reproducible with it, then something else you have installed is modifying the CSS and causing this to be off.  In order to even consider adjusting this within the plugin, you'd have to show exactly what's modifying the css and how using devtools.

ochm, 2020-01-09:
I tried the Discorded, but the same output. In screenshot is custom theme settings, no other modifications are made. I'm sorry, i don't understand about dev tools.
[image]

scottrini, 2020-01-10:
So yeah, it's your css customizations that are throwing off the styling of the bed view.  It's made specifically for the fixed width container octoprint defaults to.

You just mean it's not filling the screen without those customizations?  octoprint is made to be viewable on smaller resolutions, so it's not made to expand to large screens.

ochm, 2020-01-10:
Well, i don't know what is smaller resolution, 1920x1080 is small enough for me. :)
The plugin doesn't accept wider screen settings (and for me it is bug), and rest of Octoprint screens is rendered OK. So I will stay on wider setting of theme, which is more important for me than bed view in this plugin.

closed 2020-01-10 by ochm

scottrini, 2020-01-10:
This is not a bug as it's designed for octoprint.  You have to customize octoprint in order for it to expand to your screen size and therefore it's an issue you introduce with your modifications. So yes you're right to close this as it's not an issue with the plugin. 

1080p is pretty small for desktop, I only use ultra wide or 4k, but it's large for mobile.