scottrini/OctoPrint-PrusaLevelingGuide #9: Consider adding the directional arrows to the raw view

Kelly12612, 2020-01-24:
Just a suggestion, but it would be handy (at least I think it would be!) if the raw value screen showed the rotation arrows the way that the other views do.  I usually use the raw values with a gauge I printed from thingiverse, and I frequently turn the screw the wrong direction (dyslexics of the world untie!).
Thanks, and I really like this plugin, makes things much easier.

scottrini, 2020-01-24:
Sure, I guess I could add it in the raw values screen.  I purposely didn't at first because that's the screen in which I actually display the negative sign, which is what correlates to the direction.   I could still display it for reference.

I'm going to update this soon anyway to make it poll for the mesh bed leveling to complete faster (sometimes when I use it, it takes some time to check if it's completed), so I'm going to improve that and I'll add the arrows to the raw values display.

scottrini, 2020-01-24:
Added in 1.0.6

closed 2020-01-24 by scottrini