scottrini/OctoPrint-PrusaLevelingGuide #10: PrusaLevelingGuide stalls Octoprint Boot Process

r-a-i, 2020-01-29:
I restarted Octoprint after installing some plugging and the thing stalled for minutes. This is running on a RP-zero inside a Prusa MK3. When looking at the octoprint.log I see dozens of messages like the following:

2020-01-28 23:53:04,477 - octoprint.util.comm - ERROR - Error while processing hook PrusaLevelingGuide:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2814, in _readline
    ret = hook(self, ret)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrusaLevelingGuide/__init__.py", line 114, in mesh_level_check
    if not self.waiting_for_response == True:
AttributeError: 'PrusaLevelingGuidePlugin' object has no attribute 'waiting_for_response'
2020-01-28 23:53:09,529 - octoprint.util.comm - ERROR - Error while processing hook PrusaLevelingGuide:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2814, in _readline
    ret = hook(self, ret)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrusaLevelingGuide/__init__.py", line 114, in mesh_level_check
    if not self.waiting_for_response == True:
AttributeError: 'PrusaLevelingGuidePlugin' object has no attribute 'waiting_for_response'
2020-01-28 23:53:14,574 - octoprint.util.comm - ERROR - Error while processing hook PrusaLevelingGuide:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2814, in _readline
    ret = hook(self, ret)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrusaLevelingGuide/__init__.py", line 114, in mesh_level_check
    if not self.waiting_for_response == True:
AttributeError: 'PrusaLevelingGuidePlugin' object has no attribute 'waiting_for_response'

scottrini, 2020-01-29:
That's really strange, I have it running on all 3 of my pi's and there aren't any errors.  Only thing I could think of is if it's some really old version of octoprint or something like that, but all of my logs are clean and I can't find anyone else with this issue.  Can you provide the version octoprint and python you're using?

If you login via ssh and run ~/.oprint/bin/python --version what do you get?

r-a-i, 2020-01-29:
Thanks for looking into this!

Octoprint says Version 1.3.12 in 'about' info. 

pi@octopi:~ $ ~/oprint/bin/python --version
Python 2.7.13

If there is a required Python or Octoprint version it would be nice for the installer to assert.

scottrini, 2020-01-29:
There's no required version, it even works with the experimental 1.4 branch of octoprint with python 3 support.  

It's just an odd issue because there's so many instances running with this plugin with no error messages, so a really odd version was the only thing I could think of.  That's the version octoprint and python I have running in all 3 of my local instances without failure, so I'm really not sure what the problem could be.

Have you tried uninstalling and reinstalling it?  Can you compare your oprint/lib/python2.7/site-packages/octoprint_PrusaLevelingGuide/__init__.py  to https://github.com/scottrini/OctoPrint-PrusaLevelingGuide/blob/master/octoprint_PrusaLevelingGuide/__init__.py?

The only way for this to happen is for your instance to not fire the after startup hook, which is really odd.

scottrini, 2020-01-29:
You can try replacing your __init__.py with this version:
https://pastebin.com/7KxeaVik

Then you'll need to delete the .pyc here: `oprint/lib/python2.7/site-packages/octoprint_PrusaLevelingGuide/__init__.pyc`

If you replace this and restart without issues, let me know and I'll add that change.

closed 2020-02-19 by scottrini