scottrini/OctoPrint-PrusaLevelingGuide #11: Recreate this for non Prusa Firmware

Jakvue, 2020-02-13:
Scott,

Any chance this plugin could either be changed or branched for a Prusa not running the Prusa Marlin firmware? I’ve recently installed an SKR V1.4 board running Marlin 2.0.3, as you know the G80 and G81 commands are Prusa specific for mesh leveling and spitting out the results. For Marlin from my tests I’d need to run a G29 to perform mesh leveling and G420 V to view the results. Not super savvy with Java so I was wondering if you could help out. By the way, I'm Jack the writer of the Nylock mod guide: https://www.rearvuemirror.com/guides/nylock-mod-for-the-mk3s

scottrini, 2020-02-13:
That's a good call, I've been meaning to upgrade one of mine to an SKR 1.4 board.  I have it waiting, but haven't gone through with it yet.

I think I could _kind of_ easily add this by breaking it out into mesh leveling and mesh level checking gcode.  However, I currently use an m400 in the mesh leveling gcode to queue up the gcode commands instead of executing them one by one, so my hooks aren't checking every single gcode.  As long as I can make sure people configuring the mesh level gcode leave in that m400, or if I separate that out and hard code that (which some people will take offense to), that could definitely work.

Jakvue, 2020-02-13:
@scottrini 

So looking at your code the commands are:
`mesh_gcode = 'G28 W ; home all without mesh bed level\nM400\nG80 N3; mesh bed leveling\nG81 ; check mesh leveling results'`

New commands:
`mesh_gcode = 'G28 X 10 Y 10 ; home all without mesh bed level\nM400\nG29 N3; mesh bed leveling\nG420 V ; check mesh leveling results'`

Along with this, replacing any other mentions of G81. I'm not great at this so I'm just guessing, would that work out? I'm also currently writing up a post on the Einsy Rambo to SKR v1.4 conversion on my site, I came across a few complications that I wanted to point out and etc. Overall, it's been fun.

scottrini, 2020-02-13:
If that was it, it would just be 100% configurable right now and wouldn't need any modifications...
https://github.com/scottrini/OctoPrint-PrusaLevelingGuide/blob/master/octoprint_PrusaLevelingGuide/__init__.py#L109

When the plugin was reviewed by foosel, she didn't like watching for all gcode, so I changed it to wait for sending the value command then setting a timeout and only watching responses until that timeout's elapsed.  So that's why the setting would need to be broken out so you can specify gcode for starting the leveling and the gcode for receiving the values...

Jakvue, 2020-02-13:
@scottrini Would it be simplier to just configure the "GCODE for meshing leveling" Gcode script on the Prusa Leveling Guide plugin in OctoPrint to reflect the commands I mentioned? I haven't tried it, but if that works I think it should be fine.

scottrini, 2020-02-13:
No, that's what I was saying would not work.  That's what I meant by if that was it, it would just be configurable right now.  You'd just be able to drop that in and run with it.  But no, it needs modifications.

If you look at that line I linked, G81 is hardcoded.  It's going to wait for that specific gcode being sent to start listening for the response.  The only way for this to work is to separate the configuration into gcode for mesh leveling and gcode for retrieving the values so the code knows which gcode to listen for when querying the values.

Also, it needs to be determined if the format of the value responses is identical between the prusa fw and marlin.  I think the response would be identical or pretty close.  If there are any differences, the way it parses the values would need to be adjusted as well.

Jakvue, 2020-02-17:
@scottrini If you need help testing anything please let me know. I wish I knew how to compile plugins so I could try and test this myself but haven’t had time to look into how to do it. I’ll close the ticket for now. Also, feel free to use my Nylock guide in your read me if you want, I don’t know if you saw my comment on my page. I also just wrote a short first impression post on the SKR conversion if you’re interested. Thanks for all your work Scott.

scottrini, 2020-02-17:
Let's leave this open because I plan on making this change.  When I have some changes ready to test, I'd like to have you test and make sure it works for you before I push them.  How should I reach you?  Just @ mention you here?

Jakvue, 2020-02-17:
> Let's leave this open because I plan on making this change. When I have some changes ready to test, I'd like to have you test and make sure it works for you before I push them. How should I reach you? Just @ mention you here?

You can email me at Jack.Vue90@gmail.com or @ me here.

scottrini, 2020-02-19:
@Jakvue try this version: https://www.dropbox.com/s/x5ylbc6tdz7zypx/OctoPrint-PrusaLevelingGuide.zip?dl=0

Make sure to visit settings and specify the right gcode in the right inputs.  Let me know if it works.

Jakvue, 2020-02-19:
@scottrini Just had a chance to try it out. This is what it does:

Pre-heats to PLA
Runs Mesh 
Stops

Not sure why it doesn't output the values into the plugin or move the extruder.

Here's what it shows right after the mesh is done:
[image]

What I have in my settings:
[image]

Also, I think you meant M420 V not G420

scottrini, 2020-02-19:
Well, you have the fields configured wrong - the first field should not have the check mesh leveling results gcode.  That's the purpose of the second field.  So it couldn't work as configured right now, the M420 should be in place of that lone G29.  I actually got the G420 from your comment above from 6 days ago.  If it should be updated, I can change it to M420.   But you'll need to get those fields configured correctly before retrying.

But there's another issue here - the output of the values is different.  This has row and column headers and the prusa fw does not.  The plugin uses a regex to pull the values from the responses and it would probably still work right now, but I'd have to add another setting so it knows whether or not there are row/column headers so it knows to skip the entire first row and the first column of every line after...

Jakvue, 2020-02-19:
@scottrini My mistake. I'll make those changes on my end and run the plugin again just to see what happens. Let me know once you get the time to get the script to ignore the first column and rows and I can help test.

scottrini, 2020-02-19:
Run it as is and see if it updates - it should update, but the values should just be off.  I'll try to make time to adjust the regex later and just give you a single file to drop in.  I just wish I could test the regex myself before sending it to you!  (really I should hurry up on my SKR1.4 conversion)...

I can also understand how the UI would be a little misleading at first.  I did a copy paste of the field title.  I need to make the explanation above the fields a bit more clear about how to configure it in the future.

Jakvue, 2020-02-20:
@scottrini Alright, so I had a chance to test again after making the corrections on the Script. First Field contains:
G28
M400
G29 N3
Second Field contains:
M420 V

As of now it finishes the mesh bed leveling, spits out the M420 V Bilinear leveling Grid and the plugin sits at "Waiting for mesh level to complete" with no values displayed. Let me know if there's something I may be missing.

scottrini, 2020-02-20:
Yeah, the second option cant have the " V". It has to just be the code. If that requires the V in order to work, this approach wouldn't work like that.

It might work if you move the "M420 V" back into the first option as the last line and make the second option simply "M420"

Jakvue, 2020-02-20:
@scottrini Yeah, so it's a no go on that. Made the changes and the results are the same. It actually spits out "echo:Unknown command: ""M420 V"".

scottrini, 2020-02-20:
Wait, you copied the quotes? Of course it's not going to work with quotes.

Jakvue, 2020-02-20:
> Wait, you copied the quotes? Of course it's not going to work with quotes.

Tried without the quotes, similar results. Completes the Mesh leveling, spits out values in terminal and just sits there while the plugin says it’s waiting for mesh level to complete.

scottrini, 2020-02-21:
So thanks for trying this, but I think it's clear it'll be pretty difficult to test going back and forth like this.  What I really think is going on in your case is it's clearly sending the right commands (especially since you said you see it in terminal), but I think it's getting confused by those column/row headers.  I'd need to watch the log file and see what's coming through in there.

If you actually want to try it, you can, but I think the same gcode works on the mini and then I can try it myself.  If you want to check the log, on the pi, `tail -f ~/.octoprint/logs/octoprint.log` and that will watch the log file as you do things.  It might actually be pushing to a plugin log, but if I remember correctly it's the main octoprint.log

Jakvue, 2020-02-22:
Let me know if this helps. I wiped the logs and then ran the plugin. I did not a few errors about objects not having attributes, I'm not sure if those are relevant. 

[octoprint.log](https://github.com/scottrini/OctoPrint-PrusaLevelingGuide/files/4238802/octoprint.log)

MarcoZ76, 2020-04-20:
Can i use this plugin with my prusa Mk3S and SKR v1.4 Turbo but with Marlin 2.0.x and UBL (unified bed leveling)? I yes which start gcode i have to use? Thanks

luisleonardo, 2020-04-28:
> Can i use this plugin with my prusa Mk3S and SKR v1.4 Turbo but with Marlin 2.0.x and UBL (unified bed leveling)? I yes which start gcode i have to use? Thanks

Did you managed to make it work? I have a similar setup and was unable to find the right gcode. Thanks.

scottrini, 2020-05-19:
Unfortunately, this still isn't updated for an SKR board.  I need to work on this with my own machine to work on.  I have an SKR board, but I don't have it in any of my MK3S's yet.  One thing that worries me is that the fw is automatically converting to relative, which would have more implications for this plugin.

If anyone else wants to take a stab at this and submit a PR, feel free.

jerem2st, 2020-05-26:
> Puis-je utiliser ce plugin avec mon prusa Mk3S et SKR v1.4 Turbo mais avec Marlin 2.0.x et UBL (nivellement de lit unifié)? Je oui quel gcode de départ dois-je utiliser? Merci

> Can i use this plugin with my prusa Mk3S and SKR v1.4 Turbo but with Marlin 2.0.x and UBL (unified bed leveling)? I yes which start gcode i have to use? Thanks

the bed leveling in UBL mode has no interest here, only in ABL as shown by the mesh of Jackvue

jerem2st, 2020-05-26:
> @scottrini J'ai eu la chance de l'essayer. Voilà ce que cela fait:
> 
> Préchauffe aux arrêts de
> maille PLA Runs
> 
> Je ne sais pas pourquoi il ne sort pas les valeurs dans le plugin ou ne déplace pas l'extrudeuse.
> 
> Voici ce qu'il montre juste après le maillage:
> [image]
> 
> Ce que j'ai dans mes paramètres:
> [image]
> 
> De plus, je pense que vous vouliez dire M420 V et non G420

> @scottrini Just had a chance to try it out. This is what it does:
>
> Pre-heats to PLA
> Runs Mesh
> Stops
>
> Not sure why it doesn't output the values into the plugin or move the extruder.
>
> Here's what it [...]

Jackvue can you send me by email the sending of orders and eeprom returns in the notepad by executing the following orders:
G28
M400
G29 N3
M420 V

jeremy_110@hotmail.fr
Tks ;)

scottrini, 2021-02-16:
Completed in [1.0.13](https://github.com/scottrini/OctoPrint-PrusaLevelingGuide/releases/tag/1.0.13)

closed 2021-02-16 by scottrini

Jaw8850, 2021-02-22:
Any updates on this.  Just installed today 2/22/2021.  SKR 1.4 Turbo running Marlin.   made changes above  
G28
M400
G29 N3
M420 V
runs mesh bed level and then stops and sits "awaiting results"  am I missing something????

scottrini, 2021-02-23:
@Jaw8850 Hey, sorry about this, but the changes were actually backed out.  Too many users were having issues installing due to the new dependencies.  You can actually still install this version from here: https://github.com/acortelyou/OctoPrint-PrusaLevelingGuide
This is the fork from the developer that worked on the support for other firmwares.

We are still working to get the changes merged back in, but I'd like to get some of the other features fully tested and finalized because we will most likely block python2 in this plugin's manifest when we merge the changes back in.

james-fry, 2021-03-13:
> Any updates on this. Just installed today 2/22/2021. SKR 1.4 Turbo running Marlin. made changes above
> G28
> M400
> G29 N3
> M420 V
> runs mesh bed level and then stops and sits "awaiting results" am I missing something????

I have the same issue, even with @acortelyou fork.
I thought it was because I had ABL_BILINEAR_SUBDIVISION enabled in marlin - this makes M420 V gcode also output the CATMULL-ROM interpolated data, and I thought that was screwing up the parsing of the mesh data...
But even with ABL_BILINEAR_SUBDIVISION disabled I still cannot progress past the "awaiting results"

There are a lot of "SKR bear" users with MK52 beds and marlin firmware who would love to use this addon.