scottrini/OctoPrint-PrusaLevelingGuide #12: Prusa Mini

mvasilakis, 2020-02-19:
The plugin doesn't work with the Prusa Mini. 
Looks like G80 & G81 are not recognized.

If you decide to work on it and need someone to help you test it I'd be more than willing.
-mano

scottrini, 2020-02-19:
I didn't have any intention on supporting the mini - the last I tried to hook up an octoprint instance to my mini, it wasn't working because serial wasn't implemented via USB.  Did they eventually implement serial?

scottrini, 2020-02-19:
I'll give it a go connecting to OctoPrint again.  If those commands are supported, it's probably using the typical marlin commands/output, which I'm working on in the other issue (someone converted their MK3S to a SKR board and standard marlin).  If my changes work out there and that's the issue with the mini, it will be supported in the next release with some proper plugin configuration.

mvasilakis, 2020-02-19:
> I didn't have any intention on supporting the mini - the last I tried to hook up an octoprint instance to my mini, it wasn't working because serial wasn't implemented via USB. Did they eventually implement serial?

It's pretty much plug an play for me. USB on pi to Micro USB on Mini.

mvasilakis, 2020-02-19:
> I'll give it a go connecting to OctoPrint again. If those commands are supported, it's probably using the typical marlin commands/output, which I'm working on in the other issue (someone converted [...]

I think that will do it too. The other Bed level plugin works with the standard marlin gcodes. But it just shows the bed level is whacked and doesn't offer the level of info your plugin does.

OT: Do you know if the nyloc mod be done on the mini?

scottrini, 2020-02-19:
yeah, that's how it would work, USB to the printer's USB, but the original firmware didn't implement serial over USB, so it was just a no go.  But if it uses the standard marlin gcode, it would work with the other modification I'm working on.

I haven't seen anyone use the nylock mod yet.

mvasilakis, 2020-02-19:
With a .4mm nozzle the current (factory) level worked fine. I just threw on a vanadium .2mm nozzle and I can't get a good first layer. (Hence the reason I tried your plugin) I'm assuming the smaller the nozzle the more perfect that first layer has to be.

We have a thread about Octoprint going on PrusaPrinters.org 
https://forum.prusaprinters.org/forum/user-mods-octoprint-enclosures-nozzles/when-will-we-have-octoprint-support/paged/2/

Some trouble shooting comments there in case you run into an issue that's already been encountered.

rasssta, 2020-08-31:
Any news on this one? I ran in to the same problem with Octoprint/MINI.

LuxuSam, 2020-10-13:
Hi,
The plugin is also not working for me on the MINI.
I could imagine it is because of this line in your code as you are waiting for an response of the G81 command, but the MINI uses G29 for bed leveling.
[image]
Maybe it would work if you change it to waiting to G81 or G29
`if (gcode == "G81") or (gcode == G29):`
I tried to install the plugin on my own after changing the line, but I could not get it to work

scottrini, 2020-10-13:
@LuxuSam This plugin does not support the mini.  This issue was created asking for support for the mini, but it probably won't happen.  It's not only because of that line, the format of the values that comes back is even different, so several modifications would be required.

LuxuSam, 2020-10-13:
Oh okay, then I'm sorry for my comment

scottrini, 2021-02-16:
Completed in [1.0.13](https://github.com/scottrini/OctoPrint-PrusaLevelingGuide/releases/tag/1.0.13)

closed 2021-02-16 by scottrini