scottrini/OctoPrint-PrusaLevelingGuide #13: Error while processing hook PrusaLevelingGuide

blalor, 2020-03-22:
Seeing this in the octoprint output:

```
2020-03-22 12:53:35,268 - octoprint.util.comm - ERROR - Error while processing hook PrusaLevelingGuide:
Traceback (most recent call last):
  File "/opt/octoprint-venv/lib/python3.8/site-packages/OctoPrint-1.4.0-py3.8.egg/octoprint/util/comm.py", line 2849, in _readline
    ret = hook(self, ret)
  File "/opt/octoprint-venv/lib/python3.8/site-packages/octoprint_PrusaLevelingGuide/__init__.py", line 114, in mesh_level_check
    if not self.waiting_for_response == True:
AttributeError: 'PrusaLevelingGuidePlugin' object has no attribute 'waiting_for_response'
```

As reported by Octoprint UI:
* Octoprint Version 1.4.0
* Prusa Leveling Guide 1.0.7

scottrini, 2020-03-22:
Hey there!  Thank you for reporting this!

Someone else had previously reported this, but I can't reproduce the issue on any of the four octoprint instances I have running or in a VM, so I need help testing a fix.

I sent the previous reporter a file to drop in and test, but they never responded.  Would you be willing to drop in a file and help me test?

I'd provide a new __init__.py.  You would drop it in to `/opt/octoprint-venv/lib/python3.8/site-packages/octoprint_PrusaLevelingGuide/__init__.py`

You would also need to delete `/opt/octoprint-venv/lib/python3.8/site-packages/octoprint_PrusaLevelingGuide/__init__.pyc` if it exists (it'll be regenerated).  Then you'd restart octoprint and clear your log and see if it continues to occur.

Think you could help with this?

blalor, 2020-03-22:
Sure! Happy to!

scottrini, 2020-03-22:
https://pastebin.com/7KxeaVik

This is the one I sent the previous reporter.  It should still work now.  But let me know if this resolves that error and if so, I'll get it checked into master.

blalor, 2020-03-22:
I have a print going for a few more hours. I’ll do this next!

blalor, 2020-03-22:
That fixed the error!  The new file does have a new mix of tabs and spacing for indentation.  Thanks!

scottrini, 2020-03-22:
Awesome, thanks for testing and confirming for me!  I'll get it into master tonight and bump a new version, so within like 24hr or so Octoprint should notify you of an update to the plugin.

I'm not surprised with the mixed indentation, I crudely through that pastebin together for testing.  I'll make sure it's consistent when I push to github, though.

Thanks!

scottrini, 2020-03-22:
Fixed in 1.0.8

closed 2020-03-22 by scottrini