scottrini/OctoPrint-PrusaLevelingGuide #14: Can you add 7x7 support?

St0RM53, 2020-03-30:
It seems to only measure 3x3 grid :/ 

I know those are the screws for adjustment but the extra points can show more variation in the bed which is useful.

Kind regards

scottrini, 2020-04-02:
Hey, sorry I missed responding, but it can easily be used for 7x7.  If you go into settings and edit the mesh leveling gcode, remove the N3 from the G80.  By default, it's `G80 N3` to force 3x3 even if 7x7 is set in your firmware.  If you just delete the `N3`, it'll run whatever you have configured in your firmware.

The plugin will never show more than the current grid, but you can make this change in the settings to have it measure all 49 points.

closed 2020-04-02 by scottrini

St0RM53, 2020-04-05:
Thanks for the reply. However you should change it so it show the variation measured when 7x7 even though you can't adjust it.

scottrini, 2020-04-05:
It **will** show the measured variations when doing 7x7, but it'll just only be in the grid that matches the screws.  The plugin is specifically for the nylock method and the values for spots you can't adjust are irrelevant to leveling with the nylock mod.  You'll still get more accurate measurements if you switch it to 7x7.

St0RM53, 2020-04-06:
Ok thank you very much