scottrini/OctoPrint-PrusaLevelingGuide #16: shim edition

boelle, 2020-05-11:
@scottrini would it be possible to have an option to use precission shims also? and option for the mk42 bed?

i use that bed but not on a prusa printer so i need to use shims instead

scottrini, 2020-05-19:
I don't think this is something I'd build into this plugin, unfortunately.  You could always fork this and change it up to work for shims, but I don't feel like this would fit well into this repository.

closed 2020-05-19 by scottrini

arcady-genkin, 2021-02-19:
@boelle You can use this plugin with shims, quite easily. Just choose "Raw Values" display, and it will show you the distance in fractions of millimeters that you need to shim. This is how I'm using this plugin.
https://arcady.genkin.ca/2021/01/beer-can-mod-for-prusa-mk3-s/

scottrini, 2021-02-19:
@arcady-genkin Oh hey, that's a great use of it, I love your guide!