scottrini/OctoPrint-PrusaLevelingGuide #17: Bug: Values not updating

fwdixon, 2020-05-16:
When using the plugin, the offset values shown in the UI update after running the "begin adjustment" routine (as the should), but then do not update again until clicking "continue", which, of course, re-runs the mesh leveling and changes the values. 

A work around is to click "finish" after each adjustment, then start over, but this is cumbersome.

scottrini, 2020-05-19:
I don't think I'm completely following, I apologize.  Hopefully you can help clarify.

So you:
1. Begin adjustment -> it preheats, runs mesh leveling, gets results, moves out of the way, updates UI
2. You make the appropriate adjustments to the heat bed.
3. You click continue to re-run the mesh leveling and update the values. - UI gets updated.

Do you expect the values to automatically update between 2 and 3?  The printer can't get new values until it completely re-runs the mesh bed leveling.  So clicking continue and waiting is required.

Azelphur, 2020-05-19:
The problem I was having with it (Which I hope is the same as what OP is saying) is that the values were updating a few seconds after I pressed continue, long before mesh leveling had even started running. I worked around the problem by pressing continue twice after every screw adjustment.

scottrini, 2020-05-19:
I'm really not following that at all then.  The OP says the values **aren't** updating until clicking continue, which is expected.  You can look for an update to those values at any time by clicking the little refresh button the UI.

But the thing is, the plugin will grab new measurements whenever it can, like the Prusa Mesh Map plugin.  Any time octoprint sends a g81 and receives the measurements, it's going to update its internal values.  So that can happen separately and it can get a separate set of values.

scottrini, 2020-05-19:
Actually, did either of you guys make any changes to the default mesh leveling gcode option?

If you made any changes, can you provide what you changed it to?  That might actually be the culprit.

Azelphur, 2020-05-19:
Mine is set to 

```G28 W ; home all without mesh bed level
M400
G80 N3; mesh bed leveling
G81 ; check mesh leveling results```

which I believe is the default

scottrini, 2020-05-19:
Yeah, that's the default.  I was trying to make sure you had the M400 in there, otherwise you'd see some issues with the backend catching the measurements.

mrkag, 2020-05-29:
I was about to report the same problem, but decided to do some troubleshooting first.   I completely uninstalled Leveling Guide (deleted data), disabled all non-essential plugins, and re-installed Leveling Guide.  After this, I could not reproduce the issue so I enabled all of my installed plugins and still could not reproduce the issue.   I also tested G80 N3 vs G80 N7 and the values are now updating as expected.  How very weird.

alxlo, 2020-05-30:
I have the same faulty behaviour most of the time for quite a while. Values in the UI do not get updated after a leveling run but at the start of the next run after clicking "continue". Only then the values from the last run are shown. Clicking the refresh button does nothing at all.
i3Mk3s, OctoPrint: 1.4.0, Prusa Leveling Guide (1.0.8), Firefox 6.0.1 (64-Bit) Ubuntu

Strangely, some of the time (rarely) everything works as expected.

Mcarbuhn, 2020-07-12:
Hi, 

I also had the issue that the values only updating after "continue" the measurement. I had the Bed Level Visualizer PlugIn installed in parallel. Have now deinstalled the Bad Level Visualizer and now everything is working fine.

closed 2020-11-10 by scottrini