scottrini/OctoPrint-PrusaLevelingGuide #18: Feature Request: Increase Precision

fwdixon, 2020-05-16:
Would it possible to increase the precision of the raw values display to 3 decimal places, instead of 2?

scottrini, 2020-05-19:
I could, I could even remove rounding altogether, but it doesn't add any benefit.  I'd bet a lot of money that if I handed someone two screws or potentiometers and told them to turn one by 1.256 turns vs 1.257 turns, no one would be able to differentiate between the two.  So it really isn't helpful and then you need the additional space in the UI.

So it's perfectly doable, but I don't like this at all.
If you'd like to fork and take a stab at it and submit a PR, I'd review it.
Here's where it's rounding: https://github.com/scottrini/OctoPrint-PrusaLevelingGuide/blob/master/octoprint_PrusaLevelingGuide/__init__.py#L106
The backend rounds the values then passes them off to the frontend.  If we wanted to go all out, we'd remove rounding from the backend altogether and add an option to the frontend to adjust precision, so everyone is happy.

closed 2020-09-12 by scottrini

GrubbyZebra, 2020-09-12:
the reason for the precision in the raw measures is for those using shims and custom length spacers, not for turning screws.

scottrini, 2020-09-12:
The intention of this plugin is specifically the nylock method, so turning screws. Like I said, I don't like that for this plugin. Feel free to create your own fork. If it's configurable, you can submit a PR and I might be willing to merge it in.

scottrini, 2020-09-12:
Actually, I think I might push an update so the raw values aren't rounded at all.