scottrini/OctoPrint-PrusaLevelingGuide #20: Failure to update after firmware mod

pcweber57, 2020-07-17:
After changing the z-axis max from 210 to 205 to accommodate the copperhead hotend adapter this plug-in fails to function, hanging after passing the g80 command. 

Send: G81
Recv: Num X,Y: 7,7
Recv: Z search height: 5.00
Recv: Measured points:
Recv:   0.12500  0.17000  0.11000  0.12250  0.15500  0.14000  0.11500
Recv:   0.08500  0.10750  0.08250  0.07000  0.10500  0.11750  0.11000
Recv:   0.09000  0.12250  0.10750  0.10687  0.12750  0.13000  0.11000
Recv:   0.13000  0.10750  0.09250  0.12250  0.12000  0.10500  0.10250
Recv:   0.12750  0.14000  0.12500  0.11500  0.12500  0.13750  0.11000
Recv:   0.13000  0.14000  0.12250  0.08750  0.10750  0.12000  0.08250
Recv:   0.13250  0.16250  0.10250  0.11000  0.12250  0.12750  0.12750
Recv: ok

scottrini, 2020-07-17:
Sorry, but the fw adjustment has to be a red herring here.  All of my printers have the same fw adjustment (every one has a bondtech, including several with mosquito magnums).  They all had this when the plugin was first developed and now and it still continues to function.

edspeds, 2020-08-01:
I can confirm this is an issue, though it may be more related to 3.9.0. Background, I run two MK3's off of octoprint.  One printer is running 3.9.0 RC1 and the other is running Vertigo235's modified 3.9.0.  Prusa Leveling guide works as it should on the 3.9.0 RC1 FW but gives the same issue as PCWeber57 has on Vertigo235's release.  Below is a link to his FW that is to correct issues with Prusa's release.

https://github.com/vertigo235/Prusa-Firmware/releases/tag/MK3_3.9.1_LA15_ALTFAN_FIX_AGAIN

scottrini, 2020-08-01:
@edspeds His firmware is exactly what I run on all my MK3S's and I don't run into this problem, unfortunately.

If you'd like to debug and submit a PR for your issue, I'd certainly take a look.

edspeds, 2020-08-01:
> @edspeds His firmware is exactly what I run on all my MK3S's and I don't run into this problem, unfortunately.
> 
> If you'd like to debug and submit a PR for your issue, I'd certainly take a look.

Now I'm really confused as I run both printers off of the same instance of Octoprint, I just select a different printer to connect to.  But if we're both using the same FW then it has to be something on my end, hmmm.

scottrini, 2020-08-01:
@edspeds after I commented, I was thinking about it and realized the only difference is I build his fw myself.  I have some prints going, but I'll give one of his precompiled hexes a shot on one of the printers when it's finished.  Might as well completely rule it out.

scottrini, 2020-08-02:
Well, I was just able to reproduce it after reflashing.  Something in the changes is causing it to ignore the M400 and continue...

scottrini, 2020-08-02:
Unfortunately, this does seem to be an issue in the latest vertigo235 firmware mods.  I can switch back to the previous vertigo235 hexes I use on my printers and everything runs fine.  I've reached out to the vertigo235 maintainer to see if I can get some help figuring out the change.

edspeds, 2020-08-02:
Thanks so much for your hard work!

edspeds, 2020-08-06:
I reverted to 3.8.1 and all is well now.

scottrini, 2020-08-06:
I can even get it working on vertigo235's 3.9.0.  It's only when I pull in the 3.9.1 changes, nothing seems to work.  I don't think it's any changes he's made, I think it's new changes Prusa is working on in the 3.9.1 branch.  I did see some changes for serial functionality, but couldn't pinpoint why it would affect this at all.  What octoprint is receiving between 3.9.0 and 3.9.1 seem to be identical, but it might be how the terminal window is interpreting it.  I haven't been able to pinpoint the exact change that's causing it, I just hope it's resolved before 3.9.1 is released.

pcweber57, 2020-08-06:
I've simply done a G80 from the terminal followed by a G81. Then I grab the
output and paste it into a  file and execute gen.rb and with the output I'm
satisfactorily adjusting the nylock modified bed.

Phil

scottrini, 2020-08-06:
If I was bypassing the plugin, I'd omit generating a file and running a command and just use the relative g81 page: https://pcboy.github.io/g81_relative/

But I'd prefer to keep this thread on topic and about the plugin and fixing it with 3.9.1.

wavexx, 2020-08-17:
I was pointed here to look at this :)
I'm not sure I follow what is going on. Does the FW stalls, or does it just ignore the M400? I don't think anything changed significantly in 3.9.1 compared to 3.9.0, but I'd like to make sure we didn't break anything.

Do you have a sample gcode (the smallest you can get) I could try to run to reproduce the issue?

pcweber57, 2020-08-17:
I've not delved into the code but as soon as the leveling is complete, the extruder stops as if it was in a loop waiting for something.

wavexx, 2020-08-17:
Can you post a serial log starting from a printer reset to the moment it does this?
I could try to replay it locally.

wavexx, 2020-08-17:
Ok, I managed to reproduce this. It's a bug indeed, M400 doesn't properly handle G28 in this case.
Using a short delay (such as ``G4 S0.1``) could work-around this until it's properly fixed.

scottrini, 2020-08-17:
Oh hey, thanks!  I had been trying to poke through the code for 3.9.1 and saw some serial changes and was *thinking* that was the culprit, but clearly not.  The M400 is in place because the fw sets a timeout to wait for the G81 response.  When I was submitting my plugin to the plugins repository, foosel/Gina didn't like how I was waiting for every gcode and comparing it to the regex (which is what the PrusaMeshMap plugin does). 

The M400 is in place so there's a wait between G80 and G81.  It needs to complete the mesh bed leveling of G80 before running G81 to capture the results.  The G28 doesn't really matter, actually, and can just be removed.

wavexx, 2020-08-17:
At least currently only moves (G1) are waited-for correctly by M400.

G28/G80 might or might not be, depending on the exact time they're received on the serial (since they generate moves internally, they usually are, but M400 can still escape before they complete fully), but just looking at this I can see many other commands that don't follow this rule properly.

This was also the case before (any FW before 3.9 really). We might have moved some stuff around to make it more likely perhaps.

If you add an useless move command just before the M400 you can ensure it's always waited for, since the move will act as a barrier. It's definitely ugly, I know ;)

closed 2020-09-12 by scottrini