scottrini/OctoPrint-PrusaLevelingGuide #21: Data fields position on bed view is wrong

MauroLXX, 2020-07-22:
Hello,
could you pls check positions of output data fields on bed level view ?

Please compare linked screenshots of the same data set:
- [Bed view](https://photos.app.goo.gl/do9F9pA5odriiHah8)
- [Tabel view](https://photos.app.goo.gl/YEPzMBzdGp46FGcN9)

As you may see, field with value "0.72" has different positions in the different views. Unfortunately that causes many misunderstandings on data interpretation. I went crazy before I realized the position error :-)

My monitor is a Benq BL2411 at 1900x1200, maybe this info could help in any way...

Ciao !

scottrini, 2020-07-22:
Sorry, but there are a number of other plugins that will interfere with the bed view due to how they manipulate the default octoprint layout.  I don't have any intentions of making special changes in this plugin to accommodate how other plugins interfere with it.  There are a number of Themeify themes that will break it and some plugins that make the view "wider" that will break it as well.  You can either disable the plugins affecting it or try to fix it yourself and submit a PR.

Monitor resolution and size does not affect octoprint by default, as everything is a static width.  It only changes to a dynamic width with some additional plugins that change the way it looks, but those are the types of plugins that will interfere with bed view.

MauroLXX, 2020-07-22:
Hi Scott,
thankyou for your appreciated answed. I really didn't guess that other plugins could be the source of that weird issue.

Thankyou for your explaination !

closed 2020-07-22 by MauroLXX