scottrini/OctoPrint-PrusaLevelingGuide #24: Screw adjustment recommendations incorrect

Andy-Prusa, 2020-09-29:
Hi I’m having an issue where the recommendation for cw and ccw Rotation makes the bed level worse. It also disagrees with bed visualiser level. It also disagrees with cw and ccw recommendation using g80/g81 values inserted into https://pcboy.github.io/g81_relative/. (Which gives numbers that tie in with bed visualiser. I cannot even rotate or flip the numbers to make them fit. 
[image]
[image]
[image]

scottrini, 2020-09-29:
This plugin uses the same algorithms as the pcboy site.  You'd need to provide the raw value view along with the adjustment view in order for this to be useful for debugging.  Unfortunately, I can't reproduce any issues with the plugin using the latest version (I have it running on 4 printers).

Andy-Prusa, 2020-09-29:
I tried to show the raw values and the outputs in the attached images. The numbers are at slightly different times but I’ve made no adjustments to the bed in between.

Raw values:

0.12962  -0.12163  -0.07762  0.00940  0.10542  0.16342  0.28594
  -0.19014  -0.13065  -0.08364  -0.03715  0.07681  0.18480  0.27031
  -0.23176  -0.14774  -0.09224  -0.01886  0.06728  0.18478  0.24779
  -0.21276  -0.15027  -0.10428  -0.01330  0.07470  0.14367  0.21366
  -0.14088  -0.05437  -0.00586  0.06190  0.10114  0.15316  0.13416
  -0.03039  0.07961  0.14760  0.16560  0.16010  0.12610  0.02660
  0.09460  0.23960  0.34110  0.35010  0.27160  0.13510  -0.05790

Pcboy output:

      Raw values:
      -0.12	0.02	0.3
      -0.2	0	0.23
      0.11	0.36	-0.04
      
      Degrees:
      86°CCW	14°CW	216°CW
      144°CCW	0	166°CW
      79°CW	259°CW	29°CCW

      Fractional Turns:
      1/5CCW	0CCW	3/5CW
      2/5CCW	0	1/2CW
      1/5CW	7/10CW	1/10CCW

    
      -
Bed leveller values as in picture above
Bed visualiser as in picture above

scottrini, 2020-09-29:
I was asking for the raw values view of my plugin.

Andy-Prusa, 2020-09-29:
How do I get those?

scottrini, 2020-09-29:
Either way, I just retested locally and my values were extremely close to the pcboy site.  The only reason they're not identical is due to rounding, so I don't see an issue here.

scottrini, 2020-09-29:
By selecting raw values in the drop down.

scottrini, 2020-09-29:
https://i.imgur.com/JtVhJHw.png
https://i.imgur.com/aLe9kJc.png
Close enough, like I said, the only difference is due to rounding.

Andy-Prusa, 2020-09-29:
Ok but look at the numbers my bed leveller is outputting in the picture above. It’s nothing like you get. So a big difference.

E.g. front left your prusa levelling output  5.2 degrees 

My front left using same numbers (see picture above) 245 degrees

Andy-Prusa, 2020-09-29:
Same applies for all of them, pretty much

scottrini, 2020-09-29:
It's nothing like I get? what?  I'm using numbers from my printer, I'm not testing anything from yours.  The numbers from my printer shows the numbers are close enough between pcboy and what the plugin outputs, so this isn't something I'd look into.

Andy-Prusa, 2020-09-29:
Yes so the question was am I doing something wrong, because pcboy and bed visualiser are in agreement but this app is producing random numbers. I thought it was something you might be interested in. 🙏🖖👍

scottrini, 2020-09-29:
My screenshots show it's not producing random numbers.  If you'd like to investigate and post a PR, feel free.

closed 2020-09-29 by scottrini

Andy-Prusa, 2020-09-29:
Sorry, because you asked for my raw data, I assumed you used them in pcboy. Your output works, but the pictures I sent of mine were meant to show the disparity between your app and others. What is a PR?

scottrini, 2020-09-29:
I've never used that bed visualizer plugin, I've always used PrusaMeshMap since it's specifically for the Prusas.  But I always test against pcboy.  I was saying to send a screenshot of the raw values because obviously your theme is pushing off the location of the boxes.  It was to match up what you were getting from pcboy with what was being displayed in the plugin.  It could certainly be showing numbers in the wrong spot because your theme throwing the locations off.  You could always try table view to see if it lines up better too.

a PR is a pull request, on github.  If you fix the issue and want to contribute.

scottrini, 2020-09-29:
And the reason my values are slightly off of pcboy is because pcboy rounds the value to 2 decimal places and THEN calculates degrees (and rounds degrees down to whole number).  Mine does not round and calculates degrees using the full value.  Then degrees is also rounded to a single decimal place.  So technically mine is more accurate.  If you're using Themeify, "Discorded" is a dark theme that's compatible.  But setting custom CSS rules can also easily break the table view.

Andy-Prusa, 2020-09-29:
Ok, problem solved. I turned off themify (like you suggested it might be screwing up the table) and now it works fine.

scottrini, 2020-09-29:
Good!  Also if you switch to table view instead of bed view, you can use whatever theme you want.  But if you switch Themeify to using "Discorded," it'll be a dark theme and also be compatible.  It's just impossible to make it compatible with everything since the themes change so much.