scottrini/OctoPrint-PrusaLevelingGuide #25: Bug: Overlay not in correct position on bed

limited660, 2020-10-25:
The boxes don't position correctly over the bed image.

[image]

scottrini, 2020-10-25:
This isn't a bug, this is from your custom css from your theme. The plugin works with themeify's discorded theme and the default, but that's about it.

closed 2020-10-25 by scottrini

limited660, 2020-10-25:
This is the discorded theme...

limited660, 2020-10-25:
Also not aligned with Themeify turned off.

scottrini, 2020-10-25:
Then you either have another plugin installed that changes the css (like the one that expands the view for wide screen monitors) or you have some custom css defined.  The only thing that can throw these off is custom css.  On a base octoprint install without any modifications, it'll line up as expected.  You can simply just switch to the table view.