scottrini/OctoPrint-PrusaLevelingGuide #31: upgrade to 1.013

islander2013, 2021-02-16:
i get the following when trying to update help please;

++++++++++++++++++++++++++++++++++++++++++++++++++
Now updating Prusa Leveling Guide Plugin to 1.0.13
++++++++++++++++++++++++++++++++++++++++++++++++++

...

----------------------------------------
ERROR: Failed building wheel for scipy
Building wheel for scipy (setup.py): finished with status 'error'
Running setup.py clean for scipy
ERROR: Command errored out with exit status 1:
command: /home/pi/oprint/bin/python2 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-poQ8O1/scipy/setup.py'"'"'; __file__='"'"'/tmp/pip-install-poQ8O1/scipy/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' clean --all
cwd: /tmp/pip-install-poQ8O1/scipy
Complete output (9 lines):
`setup.py clean` is not supported, use one of the following instead:
- `git clean -xdf` (cleans all files)
- `git clean -Xdf` (cleans all versioned files, doesn't touch
files that aren't checked into the git repo)
Add `--force` to your command to use it anyway if you must (unsupported).
----------------------------------------
ERROR: Failed cleaning build dir for scipy
Successfully built OctoPrint-PrusaLevelingGuide
Failed to build scipy
Installing collected packages: scipy, OctoPrint-PrusaLevelingGuide
Running setup.py install for scipy: started
ERROR: Command errored out with exit status 1:
command: /home/pi/oprint/bin/python2 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-poQ8O1/scipy/setup.py'"'"'; __file__='"'"'/tmp/pip-install-poQ8O1/scipy/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-p2NnbN/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/oprint/include/site/python2.7/scipy
cwd: /tmp/pip-install-poQ8O1/scipy/
Complete output (152 lines):
Note: if you need reliable uninstall behavior, then install
with pip instead of using `setup.py install`:
- `pip install .`       (from a git repo or downloaded source
release)
- `pip install scipy`   (last SciPy release on PyPI)
Running from scipy source directory.
lapack_opt_info:
lapack_mkl_info:
customize UnixCCompiler
libraries mkl_rt not found in ['/home/pi/oprint/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/arm-linux-gnueabihf']
NOT AVAILABLE
openblas_lapack_info:
customize UnixCCompiler
customize UnixCCompiler
libraries openblas not found in ['/home/pi/oprint/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/arm-linux-gnueabihf']
NOT AVAILABLE
openblas_clapack_info:
customize UnixCCompiler
customize UnixCCompiler
libraries openblas,lapack not found in ['/home/pi/oprint/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/arm-linux-gnueabihf']
NOT AVAILABLE
atlas_3_10_threads_info:
Setting PTATLAS=ATLAS
customize UnixCCompiler
libraries lapack_atlas not found in /home/pi/oprint/lib
customize UnixCCompiler
libraries tatlas,tatlas not found in /home/pi/oprint/lib
customize UnixCCompiler
libraries lapack_atlas not found in /usr/local/lib
customize UnixCCompiler
libraries tatlas,tatlas not found in /usr/local/lib
customize UnixCCompiler
libraries lapack_atlas not found in /usr/lib
customize UnixCCompiler
libraries tatlas,tatlas not found in /usr/lib
customize UnixCCompiler
libraries lapack_atlas not found in /usr/lib/arm-linux-gnueabihf
customize UnixCCompiler
libraries tatlas,tatlas not found in /usr/lib/arm-linux-gnueabihf
<class 'numpy.distutils.system_info.atlas_3_10_threads_info'>
NOT AVAILABLE
atlas_3_10_info:
customize UnixCCompiler
libraries lapack_atlas not found in /home/pi/oprint/lib
customize UnixCCompiler
libraries satlas,satlas not found in /home/pi/oprint/lib
customize UnixCCompiler
libraries lapack_atlas not found in /usr/local/lib
customize UnixCCompiler
libraries satlas,satlas not found in /usr/local/lib
customize UnixCCompiler
libraries lapack_atlas not found in /usr/lib
customize UnixCCompiler
libraries satlas,satlas not found in /usr/lib
customize UnixCCompiler
libraries lapack_atlas not found in /usr/lib/arm-linux-gnueabihf
customize UnixCCompiler
libraries satlas,satlas not found in /usr/lib/arm-linux-gnueabihf
<class 'numpy.distutils.system_info.atlas_3_10_info'>
NOT AVAILABLE
atlas_threads_info:
Setting PTATLAS=ATLAS
customize UnixCCompiler
libraries lapack_atlas not found in /home/pi/oprint/lib
customize UnixCCompiler
libraries ptf77blas,ptcblas,atlas not found in /home/pi/oprint/lib
customize UnixCCompiler
libraries lapack_atlas not found in /usr/local/lib
customize UnixCCompiler
libraries ptf77blas,ptcblas,atlas not found in /usr/local/lib
customize UnixCCompiler
libraries lapack_atlas not found in /usr/lib
customize UnixCCompiler
libraries ptf77blas,ptcblas,atlas not found in /usr/lib
customize UnixCCompiler
libraries lapack_atlas not found in /usr/lib/arm-linux-gnueabihf
customize UnixCCompiler
libraries ptf77blas,ptcblas,atlas not found in /usr/lib/arm-linux-gnueabihf
<class 'numpy.distutils.system_info.atlas_threads_info'>
NOT AVAILABLE
atlas_info:
customize UnixCCompiler
libraries lapack_atlas not found in /home/pi/oprint/lib
customize UnixCCompiler
libraries f77blas,cblas,atlas not found in /home/pi/oprint/lib
customize UnixCCompiler
libraries lapack_atlas not found in /usr/local/lib
customize UnixCCompiler
libraries f77blas,cblas,atlas not found in /usr/local/lib
customize UnixCCompiler
libraries lapack_atlas not found in /usr/lib
customize UnixCCompiler
libraries f77blas,cblas,atlas not found in /usr/lib
customize UnixCCompiler
libraries lapack_atlas not found in /usr/lib/arm-linux-gnueabihf
customize UnixCCompiler
libraries f77blas,cblas,atlas not found in /usr/lib/arm-linux-gnueabihf
<class 'numpy.distutils.system_info.atlas_info'>
NOT AVAILABLE
accelerate_info:
NOT AVAILABLE
/home/pi/oprint/local/lib/python2.7/site-packages/numpy/distutils/system_info.py:639: UserWarning:
Atlas (http://math-atlas.sourceforge.net/) libraries not found.
Directories to search for the libraries can be specified in the
numpy/distutils/site.cfg file (section [atlas]) or by setting
the ATLAS environment variable.
self.calc_info()
lapack_info:
customize UnixCCompiler
libraries lapack not found in ['/home/pi/oprint/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/arm-linux-gnueabihf']
NOT AVAILABLE
/home/pi/oprint/local/lib/python2.7/site-packages/numpy/distutils/system_info.py:639: UserWarning:
Lapack (http://www.netlib.org/lapack/) libraries not found.
Directories to search for the libraries can be specified in the
numpy/distutils/site.cfg file (section [lapack]) or by setting
the LAPACK environment variable.
self.calc_info()
lapack_src_info:
NOT AVAILABLE
/home/pi/oprint/local/lib/python2.7/site-packages/numpy/distutils/system_info.py:639: UserWarning:
Lapack (http://www.netlib.org/lapack/) sources not found.
Directories to search for the sources can be specified in the
numpy/distutils/site.cfg file (section [lapack_src]) or by setting
the LAPACK_SRC environment variable.
self.calc_info()
NOT AVAILABLE
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/tmp/pip-install-poQ8O1/scipy/setup.py", line 492, in <module>
setup_package()
File "/tmp/pip-install-poQ8O1/scipy/setup.py", line 488, in setup_package
setup(**metadata)
File "/home/pi/oprint/local/lib/python2.7/site-packages/numpy/distutils/core.py", line 137, in setup
config = configuration()
File "/tmp/pip-install-poQ8O1/scipy/setup.py", line 395, in configuration
raise NotFoundError(msg)
numpy.distutils.system_info.NotFoundError: No lapack/blas resources found.
----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/oprint/bin/python2 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-poQ8O1/scipy/setup.py'"'"'; __file__='"'"'/tmp/pip-install-poQ8O1/scipy/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-p2NnbN/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/oprint/include/site/python2.7/scipy Check the logs for full command output.
Running setup.py install for scipy: finished with status 'error'
The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.``

jneilliii, 2021-02-16:
Both the scipi and numpy dependencies in a python 2.7 environment are breaking here I think. The latest version of these libraries that are supported by Python 2.7 are listed below and could be manually installed ahead of upgrade with the following commands via SSH.

```
~/oprint/bin/pip install numpy==1.16.6
~/oprint/bin/pip install scipy==1.2.3
```

The other options would be upgrading your OctoPrint virtual environment using the upgrade script found [here](https://github.com/cp2004/Octoprint-Upgrade-To-Py3) or making a backup and saving it, flashing octopi 0.18 and restoring the backup.

islander2013, 2021-02-16:
yup that fixed the issue, thank you!

closed 2021-02-16 by islander2013

spiff72, 2021-02-16:
I am having the same issue with the update failure.  I tried running the pip install commands but the scipy one fails.  

I think I will just take the backup - flash octopi 0.18 - restore backup route here.

timur-tabi, 2021-02-16:
Attempting to install scipy fails:

Collecting scipy==1.2.3
  Downloading https://files.pythonhosted.org/packages/62/4f/7e95c5000c411164d5ca6f55ac54cda5d200a3b6719dafd215ee0bd61578/scipy-1.2.3.tar.gz (23.3MB)
    100% |████████████████████████████████| 23.3MB 13kB/s 
Building wheels for collected packages: scipy
  Running setup.py bdist_wheel for scipy ... error
  Complete output from command /home/pi/oprint/bin/python2 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-r6EiLA/scipy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpxXd8fApip-wheel- --python-tag cp27:
  Running from scipy source directory.
  lapack_opt_info:
  lapack_mkl_info:
  customize UnixCCompiler
    libraries mkl_rt not found in ['/home/pi/oprint/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/arm-linux-gnueabihf']
    NOT AVAILABLE
  
  openblas_lapack_info:
  customize UnixCCompiler
  customize UnixCCompiler
    libraries openblas not found in ['/home/pi/oprint/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/arm-linux-gnueabihf']
    NOT AVAILABLE
  
  openblas_clapack_info:
  customize UnixCCompiler
  customize UnixCCompiler
    libraries openblas,lapack not found in ['/home/pi/oprint/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/arm-linux-gnueabihf']
    NOT AVAILABLE
  
  atlas_3_10_threads_info:
  Setting PTATLAS=ATLAS
  customize UnixCCompiler
    libraries lapack_atlas not found in /home/pi/oprint/lib
  customize UnixCCompiler
    libraries tatlas,tatlas not found in /home/pi/oprint/lib
  customize UnixCCompiler
    libraries lapack_atlas not found in /usr/local/lib
  customize UnixCCompiler
    libraries tatlas,tatlas not found in /usr/local/lib
  customize UnixCCompiler
    libraries lapack_atlas not found in /usr/lib
  customize UnixCCompiler
    libraries tatlas,tatlas not found in /usr/lib
  customize UnixCCompiler
    libraries lapack_atlas not found in /usr/lib/arm-linux-gnueabihf
  customize UnixCCompiler
    libraries tatlas,tatlas not found in /usr/lib/arm-linux-gnueabihf
  <class 'numpy.distutils.system_info.atlas_3_10_threads_info'>
    NOT AVAILABLE
  
  atlas_3_10_info:
  customize UnixCCompiler
    libraries lapack_atlas not found in /home/pi/oprint/lib
  customize UnixCCompiler
    libraries satlas,satlas not found in /home/pi/oprint/lib
  customize UnixCCompiler
    libraries lapack_atlas not found in /usr/local/lib
  customize UnixCCompiler
    libraries satlas,satlas not found in /usr/local/lib
  customize UnixCCompiler
    libraries lapack_atlas not found in /usr/lib
  customize UnixCCompiler
    libraries satlas,satlas not found in /usr/lib
  customize UnixCCompiler
    libraries lapack_atlas not found in /usr/lib/arm-linux-gnueabihf
  customize UnixCCompiler
    libraries satlas,satlas not found in /usr/lib/arm-linux-gnueabihf
  <class 'numpy.distutils.system_info.atlas_3_10_info'>
    NOT AVAILABLE
  
  atlas_threads_info:
  Setting PTATLAS=ATLAS
[... 234 lines cut ...]
    
    /home/pi/oprint/local/lib/python2.7/site-packages/numpy/distutils/system_info.py:639: UserWarning:
        Lapack (http://www.netlib.org/lapack/) sources not found.
        Directories to search for the sources can be specified in the
        numpy/distutils/site.cfg file (section [lapack_src]) or by setting
        the LAPACK_SRC environment variable.
      self.calc_info()
      NOT AVAILABLE
    
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-r6EiLA/scipy/setup.py", line 492, in <module>
        setup_package()
      File "/tmp/pip-build-r6EiLA/scipy/setup.py", line 488, in setup_package
        setup(**metadata)
      File "/home/pi/oprint/local/lib/python2.7/site-packages/numpy/distutils/core.py", line 137, in setup
        config = configuration()
      File "/tmp/pip-build-r6EiLA/scipy/setup.py", line 395, in configuration
        raise NotFoundError(msg)
    numpy.distutils.system_info.NotFoundError: No lapack/blas resources found.
    
    ----------------------------------------
Command "/home/pi/oprint/bin/python2 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-r6EiLA/scipy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-rCf5sh-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/oprint/include/site/python2.7/scipy" failed with error code 1 in /tmp/pip-build-r6EiLA/scipy/
You are using pip version 9.0.3, however version 21.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

timur-tabi, 2021-02-17:
I think my problem is that my OctoPi is still at version 0.15.1, which is apparently quite old now.  I'm just going to uninstall this plugin for now.

jneilliii, 2021-02-17:
Definitely the age of your octopi instance is an issue here. You could always install the previous version and then just ignore the upgrade. You can copy/paste the URL below in plugin manager > get more > ...from URL and click Install. 

```
https://github.com/scottrini/OctoPrint-PrusaLevelingGuide/releases/download/1.0.12/OctoPrint-PrusaLevelingGuide-1.0.12.zip
```