scottrini/OctoPrint-PrusaLevelingGuide #32: Can't install prusa leveling guide

jec0709, 2021-02-16:
Can't install prusa leveling guide
What should I do?
code : 
Updating, please wait.
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/scottrini/OctoPrint-PrusaLevelingGuide/archive/1.0.13.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/scottrini/OctoPrint-PrusaLevelingGuide/archive/1.0.13.zip
Downloading https://github.com/scottrini/OctoPrint-PrusaLevelingGuide/archive/1.0.13.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PrusaLevelingGuide==1.0.13) (1.5.3)
Requirement already satisfied: regex in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PrusaLevelingGuide==1.0.13) (2019.8.19)
Requirement already satisfied: numpy in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PrusaLevelingGuide==1.0.13) (1.16.6)
Collecting scipy
Downloading scipy-1.2.3.tar.gz (23.3 MB)
Requirement already satisfied: OctoPrint-FileCheck>=2020.08.07 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (2020.8.7)
Requirement already satisfied: futures<4,>=3.3; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (3.3.0)
Requirement already satisfied: semantic-version<3,>=2.8.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (2.8.5)
Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (1.1.0)
Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (0.18.2)
Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (1.4)
Requirement already satisfied: unidecode<0.05,>=0.04.14 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (0.4.21)
Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (1.0.1)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (2.0)
Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (5.3.1)
Requirement already satisfied: pylru<2,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (1.2.0)
Requirement already satisfied: pyserial<4,>=3.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (3.5)
Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (1.5.0.1)
Requirement already satisfied: wrapt<2,>=1.12.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (1.12.1)
Requirement already satisfied: chainmap<2,>=1.0.3; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (1.0.3)
Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (7.1.2)
Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (1.1.1)
Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (3.1.1)
Requirement already satisfied: netifaces<1,>=0.10.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (0.10.9)
Requirement already satisfied: scandir<2,>=1.10; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (1.10.0)
Requirement already satisfied: rsa==4.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (4.0)
Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (1.0.7)
Requirement already satisfied: tornado==5.1.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (5.1.1)
Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (0.1.5.post0)
Requirement already satisfied: feedparser<6,>=5.2.1; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (5.2.1)
Requirement already satisfied: cachelib<1,>=0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (0.1)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2020.09.23 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (2021.2.4)
Requirement already satisfied: typing<4,>=3.7.4.1; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (3.7.4.1)
Requirement already satisfied: flask<2,>=1.1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (1.1.2)
Requirement already satisfied: sentry-sdk<1,>=0.15.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (0.16.3)
Requirement already satisfied: watchdog==0.10.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (0.10.4)
Requirement already satisfied: frozendict<2,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (1.2)
Requirement already satisfied: netaddr<1,>=0.7.19 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (0.7.19)
Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (2.11.2)
Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (0.5.0)
Requirement already satisfied: emoji<1,>=0.5.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (0.5.4)
Requirement already satisfied: requests<3,>=2.23.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (2.24.0)
Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (1.0.0)
Requirement already satisfied: websocket-client<1,>=0.57 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (0.57.0)
Requirement already satisfied: monotonic<2,>=1.5; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (1.5)
Requirement already satisfied: enum34<1.2,>=1.1.10; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (1.1.10)
Requirement already satisfied: psutil<6,>=5.7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (5.7.2)
Requirement already satisfied: webassets>=2.0 in /home/pi/oprint/lib/python2.7/site-packages (from Flask-Assets<3,>=2.0->OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (2.0)
Requirement already satisfied: setuptools>=36 in /home/pi/oprint/lib/python2.7/site-packages (from markdown<3.2,>=3.1->OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (41.5.1)
Requirement already satisfied: pyasn1>=0.1.3 in /home/pi/oprint/lib/python2.7/site-packages (from rsa==4.0->OctoPrint->OctoPrint-PrusaLevelingGuide==1.0.13) (0.4.7)
[... cut ...] \r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-qiWe3X/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/oprint/include/site/python2.7/scipy Check the logs for full command output.
Running setup.py install for scipy: finished with status 'error'
The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.

scottrini, 2021-02-16:
There's a new version that might help this, but otherwise check out this comment that should get you sorted: https://github.com/scottrini/OctoPrint-PrusaLevelingGuide/issues/31#issuecomment-779587585

ajdavids, 2021-02-17:
I'm in version 1.0.12, octopi 0.17, octoprint's latest version, and the oprint venv is already python 3, yet this plug-in fails to update to 1.0.14 because scypi won't install.

draeath, 2021-02-17:
@ajdavids in your original message I see python2 is being run. Are you *positive* you are using python3 for octoprint?

If you get over that hurdle you may run into #34 as well, if you're unlucky (I'm attempting to work on that one myself)

draeath, 2021-02-17:
Tested successfully with:
    numpy==1.20.1
    scipy==1.6.0

Given your code doesn't seem very version dependent, I'm going to put the following in my PR:

    numpy~=1.20.0
    scipy~=1.6.0
 
 This spec should allow anything within 1.20 and 1.6 respectively (ie 1.6.5 would pass but not 1.5.4 or 1.7.3)

scottrini, 2021-02-18:
Sorry for these issues.  Some changes submitted that I merged added two new dependencies and many users have been having trouble installing them.  So I've rolled back the changes.  If you upgrade to 1.0.15, it's basically identical to 1.0.12 (before the changes), so just upgrade to 1.0.15 and you'll be fine.

I won't merge those changes back in unless we find a way to refactor the code and remove those dependencies.

closed 2021-02-18 by scottrini