scottrini/OctoPrint-PrusaLevelingGuide #34: cannot install on newer systems (old scipy vs gfortran 10)

draeath, 2021-02-17:
Hello,

While investigating a failure to upgrade today, I saw that scipy was failing to install with gfortran blowing up. It turns out that scipy 1.5.0 or newer is required for a successful build with gfortran 10. This is an issue in my case as there do not appear to be wheels for this platform (raspberry pi 4, alpine linux arm64, python 3.8)

[Are these specific numpy and scipy versions](https://github.com/scottrini/OctoPrint-PrusaLevelingGuide/blob/393151d/setup.py#L36) hard requirements? Scipy 1.2.3 and Numpy 1.16.6 are over a year old. If there's no reason to stay on these old versions, it might be worth looking into bumping them?

draeath, 2021-02-17:
For context, I had initially installed the plugin on the prior alpine release and have since done an in-place upgrade (which included an upgrade to gcc suite 10)

draeath, 2021-02-17:
(I'm going to fork and try bumping these - if the plugin appears to work I will report back)

ellensp, 2021-02-17:
Also getting this, brand new 2020-12-02-octopi-buster-armhf-lite-0.18.0.img install with full apt updates

draeath, 2021-02-17:
Tested successfully with:

    numpy==1.20.1
    scipy==1.6.0

Given your code doesn't seem very version dependent, I'm going to put the following in my PR:

    numpy~=1.20.0
    scipy~=1.6.0
 
 This spec should allow anything within 1.20 and 1.6 respectively (ie 1.6.5 would pass but not 1.5.4 or 1.7.3)

draeath, 2021-02-17:
@ellensp meanwhile you should be able to use `pip install git+https://github.com/draeath/OctoPrint-PrusaLevelingGuide.git` in your venv for now, though I'll be removing my repo if the PR is accepted and merged in.

Expect scipy to take *ages* to install, especially if your Pi (if you're using one, I'm assuming) doesn't have active cooling or at least passive with an open case.

Prutsium, 2021-02-17:
Oh boy ..... install fails 

Building wheel for OctoPrint-PrusaLevelingGuide (setup.py): finished with status 'done'Created wheel for OctoPrint-PrusaLevelingGuide: filename=OctoPrint_PrusaLevelingGuide-1.0.14-py3-none-any.whl size=228861 sha256=60f05ebb6c22f146eafb05669a41cacd39a4ec29ffd97c1bc3727efcc9c6c597Stored in directory: /tmp/pip-ephem-wheel-cache-6mugjsxm/wheels/6f/2b/13/8b3935c0e55186c92b5f84f320bbadc16efb9a740a0a60911fBuilding wheel for numpy (setup.py): startedBuilding wheel for numpy (setup.py): still running...Building wheel for numpy (setup.py): still running...Building wheel for numpy (setup.py): still running...Building wheel for numpy (setup.py): still running...Building wheel for numpy (setup.py): still running...Building wheel for numpy (setup.py): still running...Building wheel for numpy (setup.py): still running...Building wheel for numpy (setup.py): finished with status 'done'Created wheel for numpy: filename=numpy-1.16.6-cp37-cp37m-linux_aarch64.whl size=11887823 sha256=dc6007d22a121ec675b1e239fa7f8af78cf073d4e784cf3a3b136bae07b75ecbStored in directory: /tmp/pip-ephem-wheel-cache-6mugjsxm/wheels/11/be/f8/f44fbba1782f5914b756a602a38d8fc0cbffac1897910cc41fBuilding wheel for scipy (setup.py): startedBuilding wheel for scipy (setup.py): finished with status 'error'Running setup.py clean for scipyERROR: Command errored out with exit status 1:command: /home/pi/oprint/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-jmkhvoic/scipy/setup.py'"'"'; __file__='"'"'/tmp/pip-install-jmkhvoic/scipy/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-ewvg47ugcwd: /tmp/pip-install-jmkhvoic/scipy/Complete output (7 lines):Traceback (most recent call last):File "<string>", line 1, in <module>File "/tmp/pip-install-jmkhvoic/scipy/setup.py", line 492, in <module>setup_package()File "/tmp/pip-install-jmkhvoic/scipy/setup.py", line 468, in setup_packagefrom numpy.distutils.core import setupModuleNotFoundError: No module named 'numpy'----------------------------------------ERROR: Failed building wheel for scipyERROR: Command errored out with exit status 1:command: /home/pi/oprint/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-jmkhvoic/scipy/setup.py'"'"'; __file__='"'"'/tmp/pip-install-jmkhvoic/scipy/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' clean --allcwd: /tmp/pip-install-jmkhvoic/scipyComplete output (9 lines):`setup.py clean` is not supported, use one of the following instead:- `git clean -xdf` (cleans all files)- `git clean -Xdf` (cleans all versioned files, doesn't touchfiles that aren't checked into the git repo)Add `--force` to your command to use it anyway if you must (unsupported).----------------------------------------ERROR: Failed cleaning build dir for scipyWARNING: Error parsing requirements for numpy: [Errno 2] No such file or directory: '/home/pi/oprint/lib/python3.7/site-packages/numpy-1.19.2.dist-info/METADATA'Successfully built OctoPrint-PrusaLevelingGuide numpyFailed to build scipyInstalling collected packages: numpy, scipy, OctoPrint-PrusaLevelingGuideAttempting uninstall: numpyFound existing installation: numpy 1.19.2Can't uninstall 'numpy'. No files were found to uninstall.Running setup.py install for scipy: startedERROR: Command errored out with exit status 1:command: /home/pi/oprint/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-jmkhvoic/scipy/setup.py'"'"'; __file__='"'"'/tmp/pip-install-jmkhvoic/scipy/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-uu2tprbp/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/oprint/include/site/python3.7/scipycwd: /tmp/pip-install-jmkhvoic/scipy/Complete output (168 lines):Note: if you need reliable uninstall behavior, then installwith pip instead of using `setup.py install`:- `pip install .`       (from a git repo or downloaded sourcerelease)- `pip install scipy`   (last SciPy release on PyPI)Running from scipy source directory.lapack_opt_info:lapack_mkl_info:customize UnixCCompilerlibraries mkl_rt not found in ['/home/pi/oprint/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/aarch64-linux-gnu']NOT AVAILABLEopenblas_lapack_info:customize UnixCCompilercustomize UnixCCompilerlibraries openblas not found in ['/home/pi/oprint/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/aarch64-linux-gnu']NOT AVAILABLEopenblas_clapack_info:customize UnixCCompilercustomize UnixCCompilerlibraries openblas,lapack not found in ['/home/pi/oprint/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/aarch64-linux-gnu']NOT AVAILABLEatlas_3_10_threads_info:Setting PTATLAS=ATLAScustomize UnixCCompilerlibraries lapack_atlas not found in /home/pi/oprint/libcustomize UnixCCompilerlibraries tatlas,tatlas not found in /home/pi/oprint/libcustomize UnixCCompilerlibraries lapack_atlas not found in /usr/local/libcustomize UnixCCompilerlibraries tatlas,tatlas not found in /usr/local/libcustomize UnixCCompilerlibraries lapack_atlas not found in /usr/libcustomize UnixCCompilerlibraries ta [... cut ...] "'"'; __file__='"'"'/tmp/pip-install-jmkhvoic/scipy/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-uu2tprbp/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/oprint/include/site/python3.7/scipy Check the logs for full command output.Running setup.py install for scipy: finished with status 'error'

ellensp, 2021-02-17:
@draeath  I can confirm your fork compiles and seems to work as expected

scottrini, 2021-02-17:
@acortelyou Unfortunately, I'm getting close to the point of backing out your recent changes because of the new dependencies.  The issues keep coming in for exactly what I was worried about - a lot of people are having issues getting these installed and it's not something I intend to provide support for going forward.

draeath, 2021-02-17:
Closed PR #34 as either of the likely paths forward would make it irrelevant. See [this comment](https://github.com/scottrini/OctoPrint-PrusaLevelingGuide/pull/35#issuecomment-780700579) for discussion, if you like.

draeath, 2021-02-18:
Fixed by d960e50

closed 2021-02-18 by draeath