scottrini/OctoPrint-PrusaLevelingGuide #36: Cannot update get an error 401 on http://newstream.getanywhere.io/api/dev_settings

MkLHX, 2021-02-17:
Hi, trying to update this plugin on mk2s prusa and get an error about api url.

i'm using octoprint 1.5.3 on raspberry pi 3A+

```2021-02-17 09:59:45,738 - octoprint.plugins.softwareupdate - INFO - Starting update of PrusaLevelingGuide to 1.0.14...
2021-02-17 09:59:46,885 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2021-02-17 09:59:46,888 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2021-02-17 09:59:46,888 - octoprint.util.pip - INFO - ==> pip ok -> yes
2021-02-17 09:59:57,357 - backoff - INFO - Backing off __get_dev_settings__(...) for 28.7s (HTTPError: 401 Client Error: Unauthorized for url: http://newstream.getanywhere.io/api/dev_settings)
2021-02-17 10:00:17,653 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-17 10:00:26,277 - backoff - INFO - Backing off __get_dev_settings__(...) for 109.2s (HTTPError: 401 Client Error: Unauthorized for url: http://newstream.getanywhere.io/api/dev_settings)
2021-02-17 10:01:37,831 - octoprint.plugins.softwareupdate - ERROR - Update of PrusaLevelingGuide can not be performed, please also check plugin_softwareupdate_console.log for possible causes of this
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 1788, in _perform_update
    target, populated_check, target_version, log_cb=self._log, online=online
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/updaters/pip.py", line 144, in perform_update
    "Error while executing pip install", (stdout, stderr)
UpdateError: Error while executing pip install
2021-02-17 10:02:15,830 - backoff - INFO - Backing off __get_dev_settings__(...) for 57.2s (HTTPError: 401 Client Error: Unauthorized for url: http://newstream.getanywhere.io/api/dev_settings)
2021-02-17 10:03:13,373 - backoff - INFO - Backing off __get_dev_settings__(...) for 136.0s (HTTPError: 401 Client Error: Unauthorized for url: http://newstream.getanywhere.io/api/dev_settings)
2021-02-17 10:05:29,712 - backoff - INFO - Backing off __get_dev_settings__(...) for 146.3s (HTTPError: 401 Client Error: Unauthorized for url: http://newstream.getanywhere.io/api/dev_settings)
2021-02-17 10:07:56,496 - backoff - INFO - Backing off __get_dev_settings__(...) for 148.8s (HTTPError: 401 Client Error: Unauthorized for url: http://newstream.getanywhere.io/api/dev_settings)
2021-02-17 10:10:25,624 - backoff - INFO - Backing off __get_dev_settings__(...) for 247.5s (HTTPError: 401 Client Error: Unauthorized for url: http://newstream.getanywhere.io/api/dev_settings)
2021-02-17 10:10:31,890 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.1.114
2021-02-17 10:10:32,457 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.1.114
2021-02-17 10:10:32,521 - octoprint.server.util.flask - INFO - Passively logging in user cybergogo from ::ffff:192.168.1.114
2021-02-17 10:10:32,523 - octoprint.access.users - INFO - Logged in user: cybergogo
2021-02-17 10:10:32,595 - octoprint.server.util.sockjs - INFO - User cybergogo logged in on the socket from client ::ffff:192.168.1.114
2021-02-17 10:14:33,507 - backoff - INFO - Backing off __get_dev_settings__(...) for 192.7s (HTTPError: 401 Client Error: Unauthorized for url: http://newstream.getanywhere.io/api/dev_settings)
2021-02-17 10:15:17,654 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-17 10:16:25,875 - octoprint.plugins.softwareupdate - INFO - Starting update of PrusaLevelingGuide to 1.0.14...
2021-02-17 10:17:46,560 - backoff - INFO - Backing off __get_dev_settings__(...) for 47.1s (HTTPError: 401 Client Error: Unauthorized for url: http://newstream.getanywhere.io/api/dev_settings)
2021-02-17 10:17:46,913 - octoprint.plugins.softwareupdate - ERROR - Update of PrusaLevelingGuide can not be performed, please also check plugin_softwareupdate_console.log for possible causes of this
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 1788, in _perform_update
    target, populated_check, target_version, log_cb=self._log, online=online
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/updaters/pip.py", line 144, in perform_update
    "Error while executing pip install", (stdout, stderr)
UpdateError: Error while executing pip install
2021-02-17 10:18:33,991 - backoff - INFO - Backing off __get_dev_settings__(...) for 90.1s (HTTPError: 401 Client Error: Unauthorized for url: http://newstream.getanywhere.io/api/dev_settings)
2021-02-17 10:20:04,402 - backoff - INFO - Backing off __get_dev_settings__(...) for 72.9s (HTTPError: 401 Client Error: Unauthorized for url: http://newstream.getanywhere.io/api/dev_settings)
2021-02-17 10:21:17,664 - backoff - INFO - Backing off __get_dev_settings__(...) for 206.5s (HTTPError: 401 Client Error: Unauthorized for url: http://newstream.getanywhere.io/api/dev_settings)
2021-02-17 10:21:40,377 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.1.114````

scottrini, 2021-02-17:
Those 401's for that URL are being reported from something called backoff, it's not from this pluign.  I can understand how it would look like it, though, since the leveling guide update errors are right next to those messages.

Did you by chance find anything in the other log it mentions? (plugin_softwareupdate_console.log)

MkLHX, 2021-02-17:
This is the log for the same periode from pugin_softwareupdate_console.log
```

2021-02-17 10:01:37,808 ! command: /home/pi/oprint/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-kw7tAG/scipy/setup.py'"'"'; __file__='"'"'/tmp/pip-install-kw7tAG/scipy/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-kw7tAG/scipy/pip-egg-info
2021-02-17 10:01:37,809 ! cwd: /tmp/pip-install-kw7tAG/scipy/
2021-02-17 10:01:37,810 ! Complete output (53 lines):
2021-02-17 10:01:37,810 ! Traceback (most recent call last):
2021-02-17 10:01:37,810 ! File "<string>", line 1, in <module>
2021-02-17 10:01:37,811 ! File "/tmp/pip-install-kw7tAG/scipy/setup.py", line 492, in <module>
2021-02-17 10:01:37,811 ! setup_package()
2021-02-17 10:01:37,811 ! File "/tmp/pip-install-kw7tAG/scipy/setup.py", line 488, in setup_package
2021-02-17 10:01:37,812 ! setup(**metadata)
2021-02-17 10:01:37,812 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/__init__.py", line 144, in setup
2021-02-17 10:01:37,813 ! _install_setup_requires(attrs)
2021-02-17 10:01:37,813 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/__init__.py", line 139, in _install_setup_requires
2021-02-17 10:01:37,813 ! dist.fetch_build_eggs(dist.setup_requires)
2021-02-17 10:01:37,814 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/dist.py", line 720, in fetch_build_eggs
2021-02-17 10:01:37,814 ! replace_conflicting=True,
2021-02-17 10:01:37,815 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 782, in resolve
2021-02-17 10:01:37,815 ! replace_conflicting=replace_conflicting
2021-02-17 10:01:37,815 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1065, in best_match
2021-02-17 10:01:37,816 ! return self.obtain(req, installer)
2021-02-17 10:01:37,816 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1077, in obtain
2021-02-17 10:01:37,816 ! return installer(requirement)
2021-02-17 10:01:37,817 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/dist.py", line 787, in fetch_build_egg
2021-02-17 10:01:37,817 ! return cmd.easy_install(req)
2021-02-17 10:01:37,817 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 679, in easy_install
2021-02-17 10:01:37,818 ! return self.install_item(spec, dist.location, tmpdir, deps)
2021-02-17 10:01:37,818 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 705, in install_item
2021-02-17 10:01:37,819 ! dists = self.install_eggs(spec, download, tmpdir)
2021-02-17 10:01:37,819 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 890, in install_eggs
2021-02-17 10:01:37,819 ! return self.build_and_install(setup_script, setup_base)
2021-02-17 10:01:37,820 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1158, in build_and_install
2021-02-17 10:01:37,820 ! self.run_setup(setup_script, setup_base, args)
2021-02-17 10:01:37,820 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1144, in run_setup
2021-02-17 10:01:37,821 ! run_setup(setup_script, args)
2021-02-17 10:01:37,821 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 253, in run_setup
2021-02-17 10:01:37,821 ! raise
2021-02-17 10:01:37,822 ! File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
2021-02-17 10:01:37,822 ! self.gen.throw(type, value, traceback)
2021-02-17 10:01:37,823 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 195, in setup_context
2021-02-17 10:01:37,823 ! yield
2021-02-17 10:01:37,823 ! File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
2021-02-17 10:01:37,824 ! self.gen.throw(type, value, traceback)
2021-02-17 10:01:37,824 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 166, in save_modules
2021-02-17 10:01:37,824 ! saved_exc.resume()
2021-02-17 10:01:37,825 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 141, in resume
2021-02-17 10:01:37,825 ! six.reraise(type, exc, self._tb)
2021-02-17 10:01:37,826 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 154, in save_modules
2021-02-17 10:01:37,826 ! yield saved
2021-02-17 10:01:37,826 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 195, in setup_context
2021-02-17 10:01:37,827 ! yield
2021-02-17 10:01:37,827 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 250, in run_setup
2021-02-17 10:01:37,828 ! _execfile(setup_script, ns)
2021-02-17 10:01:37,828 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 45, in _execfile
2021-02-17 10:01:37,828 ! exec(code, globals, locals)
2021-02-17 10:01:37,829 ! File "/tmp/easy_install-o8zMKU/numpy-1.20.1/setup.py", line 30, in <module>
2021-02-17 10:01:37,829 ! raise RuntimeError("Python version 2.7 or >= 3.4 required.")
2021-02-17 10:01:37,829 ! RuntimeError: Python version >= 3.7 required.
[... cut ...]
2021-02-17 10:17:46,405 ! exec(code, globals, locals)
2021-02-17 10:17:46,405 ! File "/tmp/easy_install-WsyP2J/numpy-1.20.1/setup.py", line 30, in <module>
2021-02-17 10:17:46,406 ! raise RuntimeError("Python version 2.7 or >= 3.4 required.")
2021-02-17 10:17:46,406 ! RuntimeError: Python version >= 3.7 required.
2021-02-17 10:17:46,407 ! ----------------------------------------
2021-02-17 10:17:46,911 ! ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

```

ruralaccent, 2021-02-18:
I can't update the plugin either. It's upgraded every time until this update. I've used it for about 2 years now.

scottrini, 2021-02-18:
The log shows that it's having issues installing the new dependencies that were introduced, but those were already rolled back with 1.0.15.  If you install 1.0.15, it's basically 1.0.12 again, so it rolls back to prior to the new dependencies and installs fine.

closed 2021-02-18 by scottrini

MkLHX, 2021-02-18:
Today the plugin is allowed to update. I don't know about what dependencies you told but in fact everything works now.

scottrini, 2021-02-18:
When I originally wrote the plugin, it really only relied on regex, which is always included in octoprint.  Someone recently submitted some code changes to add support for the mini and other firmwares. Unfortunately, it introduced new dependencies - numpy and scipy.  A lot of users have been having issues installing those dependencies.  Your log's error is coming from numpy.

So many people were having issues, I rolled back those merged changes yesterday.  I won't be merging them back in unless they're refactored so they no longer require those dependencies.