scottrini/OctoPrint-PrusaLevelingGuide #38: Heat only the bed to the selected profile temp, not the nozzle

arcady-genkin, 2021-02-19:
Hi, first of all, thanks for providing this plugin, it's a great tool!

My biggest beef with it is that it unnecessarily heats up the nozzle to the specified profile temperature.
There is absolutely no reason to do that, and may lead to some filament being baked inside the nozzle during the adjustment process (which may take some time and many iterations).
Can you please update the plugin to only heat the table, but leave the nozzle unheated?

Thanks!

scottrini, 2021-02-19:
You can do this by just creating new temperature profiles in octoprint that only heat up the bed.  Go to Settings->Temperatures.  Scroll down and click new.  Just leave the extruder temperature at 0 and fill in a temperature for the bed, then use that profile for preheating.  And yeah, telling the plugin to preheat then disabling it in the other tab means it'll never start, it'll stay in a loop waiting for the hotend to reach that temperature and since it'll never reach the temperature, it'll never begin.  So you have to take advantage of the fact that it takes the temperature profiles straight from octoprint and create some new ones.

However, there **is** a reason for preheating the nozzle - the PINDA.  It doesn't need to heat up all the way to material temperatures, but it should be heated.  This becomes moot only once you've upgraded to a SuperPINDA since they're no longer affected by temperature.  But the PINDA is even the reason a lot of the "pretty" profiles heat the nozzle to 160 before doing the mesh bed leveling, so they'll have more accurate results.

arcady-genkin, 2021-02-19:
@scottrini 
Thanks for replying. I don't understand how heating PINDA affects the accuracy of adjustment.
From what I've seen PINDA is accurate to about 0.2 micrometers at room temperature (see this, for example https://www.youtube.com/watch?t=559&v=il9bNWn66BY). Heating it may change its sensitivity and/or accuracy.

If the sensitivity changes, that should not matter for the purposes of bed alignment, because only relative measurements are important. But change in sensitivity *is* important for printing the first layer, so, I guess, this is why preheating PINDA for printing makes a difference.

If the accuracy changes, then, I guess, it would make sense to choose the profile for best accuracy. However, 0.2 micrometers at room temperature is definitely good enough, so even if heating PINDA up increases accuracy, that should not make any practical difference.

Am I missing something?

scottrini, 2021-02-19:
Regardless of how the PINDA functions, it won't change how this plugin works - it will always just honor the temperature profiles defined in octoprint.  Some users will always prefer both to be preheated.  So if you don't want the hotend to be preheated, just define a profile that doesn't have a temperature for the hotend.

closed 2021-02-19 by scottrini

arcady-genkin, 2021-02-19:
I can see your point, although I would be concerned about people clogging their nozzles by keeping the nozzle heated for the duration of bed alignment without extruding any filament. If it were up to me, I would definitely disable nozzle heating during adjustment to avoid that, but you are the boss.

scottrini, 2021-02-19:
See I'm not sure how big of a concern that is, as none of my printers have ever clogged from idling preheated before and I've never heard of that.

But really this boils down to how it makes sense to implement.  I originally built it to just use the bed temperature, but someone asked why it didn't use both temperatures, so I made it use both.  So if I change it to ignore that temperature, someone will be upset if they can't preheat their nozzle.  I think the best course forward is the most flexibility - just honoring the profiles that exist.  So you can have it behave however you want based on the profiles you define and use.

You have inspired me that some things could be improved here, so I'm going to think about how to do this, but I think there should at least be some additional documentation around this and how it uses the profiles built in to octoprint...and how to have it just use the bed preheating.  Alternatively, maybe I could have this plugin add some default base profiles such as "bed only 50 degrees", etc...

arcady-genkin, 2021-02-19:
Alternatively have two drop-down lists something like this:

Preheat: bed <temps> nozzle <temps>

I had ASA clog my nozzle once when I left it heated up for a relatively long time. Some of it seeped out and baked onto the nozzle from the outside, and some of it got solidified inside the nozzle. I had to burn it using a blow torch (with the nozzle removed).

scottrini, 2021-02-19:
I was thinking of something similar, but adding two checkboxes.  Still list the octoprint profiles, but have checkboxes to enable the preheat of each instead of disabling preheating entirely.  Next time I'm in there, I'll play with it.  I have a bigger priority, though, I've been meaning to fix the bed display and convert it to html5 canvas so user's custom css profiles won't throw off the layout.

arcady-genkin, 2021-02-20:
Yeah, checkboxes would work just as well.
I did take your suggestion and defined bed-only temp profiles in Octoprint for now.