scottrini/OctoPrint-PrusaLevelingGuide #39: Preheat Endless Loop

NickT-in-SD, 2021-02-20:
I tried to run a pre-heated bed level using the latest version and the bed actually heated up, but it never proceeded to measure anything. When I went to the temperature tab on OctoPrint, it didn't show the actual bed and tool temperature, or targets, despite the printer showing the preheated temperatures and targets correctly.

scottrini, 2021-02-23:
Was this using the version with the checkboxes for nozzle/bed?  Or were you using the version before and just using a profile with only a temperature for the bed?

scottrini, 2021-02-23:
And when this actually happened, what status did the plugin show?

NickT-in-SD, 2021-02-23:
Hi, thank you for getting back to me. Really great plugin!

Regarding your first question: the version I was using had a single
checkbox for preheating profiles, not separate checkboxes for heating the
bed and/or the nozzle. Though I have wondered if it would be possible to
have a bed-only heat option since I end up with oozing from the nozzle and
I don't believe the nozzle needs heat for the measurement procedure, at
least not with the new Super PINDA (but I could be wrong, pretty new with
all of this)

Regarding the second question: the status of the plugin was connected /
live as far as I could tell. I was able to resolve the issue by restarting
OctoPrint.

Have you considered an option to run a 7x7 mesh measurement to generate a
mesh map similar to the deprecated mesh map plugin? That would be nice for
fine tuning.

Let me know if there's anything else you would like me to do to test it.

Thanks,
Nick

scottrini, 2021-02-23:
For the preheating - check the README.md, it shows a screenshot of how to define a temperature profile with only a bed temperature.  It was built to use the temperature profiles you define in octoprint, so you had the ability to choose what it preheated already, just by defining a profile for the desired behavior.  However, update to the latest, and you'll get checkboxes for preheating the nozzle and/or bed.  So you can select a profile and optionally disable one for the profile.

With 7x7 - you can do this currently by going into settings and editing the gcode.  If you replace `G80 N3` with `G80` it will use the mesh bed leveling you have configured for the firmware. The `N3` enforces 3x3.  Personally, I always use 3x3 first then switch to 7x7 when I start to fine tune.  I might just remove the N3 from the default settings, but no matter what, I'm going to add a section to the README about this.

The preheating issue you're describing, I can't reproduce on any of my test instances.  So I'd say to try to reproduce and create a video of it occurring so we can pinpoint what's going on if possible.  The code that goes through this process lives in the UI (the browser), so if you reload the page it will lose your place in the process.  If the plugin was trying to preheat at the time, it would've shown "Preheating" for the status.  Either way, you should update to latest first.

NickT-in-SD, 2021-02-23:
Awesome. Thanks very much for the detailed feedback. I'll do some testing
and report back.

One thing I was thinking about is whether or not I could use the
measurement values to deduce the bed level correction values in the Prusa
firmware. Since the measurement process gives me deviation in mm and the
bed level correction gives me micrometers, I should be able to translate.
But given Prusa only uses front/rear/left/right correction, maybe it won't
work that well.

Best,
Nick

closed 2021-03-02 by scottrini