scottrini/OctoPrint-PrusaLevelingGuide #40: Bed Image not shown

Prutsium, 2021-02-22:
With the latest version (but also before) my Bed image is not shown.
[image]

scottrini, 2021-02-23:
This is something I'd need more info to be able to help you with.  This is the new view that was just released, so you're saying in the other bed view it also doesn't show?

It's going to end up being one of a few things, but everything I'd need more info from developers tools in your browser.
1. It could be a browser plugin etc blocking the image - that would be shown in the console and/or network tab of developers tools.  you're looking for photo_heatbed.png
2. It could be a custom theme/css/etc that's applying some theme to img tags or causing them not to display.  The two different views load the image differently, though, so it would be odd for it to affect both.
3. It could be getting a 404 looking for photo_headbed.png if the folder structure is different for some reason, but again, I'd need more info from devtools.

Do you have any plugins installed like ublock origin etc?  If so, try disabling everything like that.  Also, what browser are you using?

This really isn't doing anything fancy to load that image, especially not in the old view.  In the new view it's drawing the image to the html canvas, so it's doing a little more with the image, but the values are successfully being drawn to the canvas for you, so it's clearly the loading of the image.

Prutsium, 2021-02-23:
> <OctoPrint-PrusaLevelingGuide@noreply.github.com>
> shown (#40)

Ok will do the things you suggested and will check if i can get more 
info from the logs (or firefox console)
Browser = Firefox
Theme= UI Customizer but also before i had the same issue without UI 
Customizer)

Will report back when done (later in the evening)

scottrini, 2021-02-23:
If we can't figure it out back and forth here, I'd be willing to get on a quick screen share to figure out what could be causing it.

Prutsium, 2021-02-23:
Yeah well the reason is simple and various plugins have / had this problem:
Running Octoprint in Multiple instances .....
You are looking for: IP/plugin/PrusaLevelingGuide/static/img/photo_heatbed.png
But when running instances we also have multiple doc roots / ports the same issue was with the telegram plugin and Gilles fixed this.
In my case i have 2 instances:
IP/red/plugin/PrusaLevelingGuide/static/img/photo_heatbed.png
IP/blue/plugin/PrusaLevelingGuide/static/img/photo_heatbed.png

But the plugin does not check it's origin so goes to IP/plugin/PrusaLevelingGuide/static/img/photo_heatbed.png

scottrini, 2021-02-23:
Ok, I get it.  It's because I'm using an absolute path starting with /plugin.  So it's ignoring whatever base path you have for the instance.  Thanks.

I think I know how to fix it, I'll test using relative and get it pushed if I can get it working.  I'll update this with the release it's fixed in.

scottrini, 2021-02-23:
Fixed in [1.0.17](https://github.com/scottrini/OctoPrint-PrusaLevelingGuide/releases/tag/1.0.17)

closed 2021-02-23 by scottrini