scottrini/OctoPrint-PrusaLevelingGuide #42: M600 after MBL

zedaxisllc, 2021-03-02:
I don't know why it's happening, but it shows up in my terminal in octoprint - it's happened on multiple machines. Right after running MBL and before the printer moves to the "adjustment position" an M600 gets called to do a filament change. Sometimes, the filament change will complete and the printer then moves to the adjustment position, but rarely the printer just stops, as if the movement commands to move to the adjustment position were either ignored or never sent. When that happens the bed height statistics are not updated and it's necessary to click 'continue' again to run the process again.

I don't have logs - I'm sorry. Next time I run the plugin, I'll turn logging on and see if I can capture anything. 

Maybe someone else has experienced this?

scottrini, 2021-03-02:
So, nothing in the code base has an M600 configured anywhere (you can search the repository and the code).  It's not in the default settings, so the only way to get this plugin to send an M600 is to change the default settings and add one into the gcode.

I'll leave this open for a few days before closing it if you want, but the only way for this plugin to send gcode is to have it configured to be sent.

zedaxisllc, 2021-03-02:
No, if it's not in the code, it's not in the code. I may be having something wonky going on with my filament sensor. I'll try it with filament actually loaded and see if still happens. 

It's a fantastic plugin by the way. Great work. Thanks for making it.

scottrini, 2021-03-02:
You 100% confirmed you don't have it configured in either of the boxes in settings, right?  There's an M400 in there, so maybe it got changed or something.

Otherwise, yeah, I'd be curious if it was the filament sensor - maybe try disabling it and see if you still see the M600 pop up.  OR I was considering it might be some combination of other plugins you might have installed.  I have no idea for sure, just the things on top of mind...

zedaxisllc, 2021-03-02:
The only two changes that I made to anything was to deselect 'preheat using profile' and change 'raw values' to 'degrees'. Everything else is set however it was configured when it was installed. All of my printers are running at the moment so I can't test anything. If you want to leave this open, I'll revisit in the morning. I really don't have very many plugins installed but I'll dig around a bit. Your curiosity has made me more curious...

closed 2022-11-28 by scottrini

HarryTutt, 2023-03-13:
I have experienced this behavior and the only M600 is checked in the octodash plugin settings
the equipment is a MK3s+ with the MMU2s with octoprint on a raspberry pi4 4gb and a hyperpixel 4.0
it was a clean install
searching I stumbled upon this annotation