scottrini/OctoPrint-PrusaLevelingGuide #43: Plugin stuck if printer resets

jmsaltzman, 2021-03-10:
If the printer resets after the user clicks "Begin Adjusting", the plugin freezes. It will work after restarting Octoprint.
It would be nice to have a "cancel" button or for it to check (and automatically cancel) if something's not right.

scottrini, 2021-03-10:
A cancel button is a fair request so you don't have to reload and it would be an easy add.  I'll throw one in the next time I'm making changes.

But you don't have to restart octoprint for this.  All of the logic and state is in the UI (client side JS), so if you just reload the tab, it'll reset.

jmsaltzman, 2021-03-11:
Thank you! Sorry, reloading it, it DOES work correctly and go back to Idle. So... I'm not sure what I was doing wrong. NEVERMIND! I'll cancel :D

jmsaltzman, 2021-03-11:
All fine after all, sorry to bug you!

closed 2021-03-11 by jmsaltzman