scottrini/OctoPrint-PrusaLevelingGuide #44: Nozzle crashed into bed on second level point

AZGLi, 2021-04-13:
I have calibrated, set the nozzle height, and leveled the bed manually as close as I can. When I run this tool, the front left bed point is measured and then all others crash the nozzle into the bed. It acts like the right side of the bed is tilted up about 10 mm. I have been unable to complete a level run. 
Printer is a Prusa MK3S, running OctoPrint 1.3.6 and Bed Leveling Guide 1.0.17 on an RPi 4B 4GB.

scottrini, 2021-04-13:
Unfortunately, there's nothing this plugin does that could cause that. The only way for this to occur is if your pinda isn't firing on those points. That's what makes the fw stop when it's checking each point.

This plugin simply sends the same gcode (G80) the printer uses to initiate mesh bed leveling, then the printer fw does the rest.

AZGLi, 2021-04-14:
You are correct, sorry! It happens with G80 no matter the source. Something else is messed up. I can print, I just can't map the bed for some reason. Thanks for the reply.

closed 2021-04-14 by AZGLi

scottrini, 2021-04-14:
No problem, but are you still having the issue?  Here's some things I'd check -
Watch the light on the back of the PINDA first and see how it's behaving (you can see when it's firing).  If it's not firing when it's going for those points, it's either the PINDA is going bad/connection is loose OR the PINDA is too high.

If it was just too high, you'd have an issue with all points.  So if it's only having an issue with certain ones, it could be when it moves to those areas, it's putting more strain on the cable bundle and causing a problem...

AZGLi, 2021-04-15:
Thanks. I hadn't thought about the PINDA failing, but with my luck with these machines it probably is. I will test again; I did check the height and it seems OK. It definitely isn't detecting the bed properly. The LCD module just failed so I have to wait for that to get replaced before I can test more, and the PINDA upgrade is out of stock so I might be stuck with my MK2.5 for a while.