scottrini/OctoPrint-PrusaLevelingGuide #45: Bed level values are not updated after mesh bed levelling

KRider92, 2021-12-27:
Hi,
I'm facing the issue that the bed level values do not update until after Auto-Home is done after scanning. This leads to the following behavior (changes in **bold** text)
- Preheat (if enabled)
- Send the mesh level command and retrieve results **(let's call this "Scan(n-1)")**
- Wait for the command to complete
- Send gcode to move the bed and extruder out of the way **("Scan(n)")**
- **Send the mesh level command and retrieve results**
- **The plugin auto-homes the printer**
- **The plugin shows the text** "Results received; make adjustments now. When ready, click continue below." **only now**
- Update the UI with the values from scan(n-1)

This is quite frustrating and leads to incorrect results.
I'm using the plugin on OctoPrint 1.7.2, Prusa Mk3s MMU2s, Firmware 3.9.3-3556, Plugin version 1.0.17.

KRider92, 2021-12-27:
Also, during the plugins runtime, the "refresh" arrow is greyed out, which means I can't force the plugin to reload the currnet mesh.

KRider92, 2021-12-27:
Just found out that the update works as expected with `G80 N3` as Levelling command. This is too inaccurate for me, though.

scottrini, 2021-12-27:
I just re-ran all of the test suites with that the version setup you provided and everything is working still.  You can feel free to edit the G80 command to exclude the N3 and let it use whatever's setup in the firmware (that's how I have all of mine configured).  Specifying the N3 won't cause it to work or not (the output from the printer is identical), but modifying some of the other GCODE in the config can cause it to not receive values.

KRider92, 2021-12-28:
Hi,
Fresh day, fresh start - I re-ran everything and found it to be working when trying to make a recording of what's happened >.<

Seems to be all good now, closing this.

closed 2021-12-28 by KRider92