scottrini/OctoPrint-PrusaLevelingGuide #46: Feature Request: Averaging functionality

twes100, 2022-02-15:
The plugin works really nicely, but in practice the levelling process is tedious to run simply because of the variability in the results that are returned by the printer's bed levelling function.

Some people resort to making multiple MBL runs and averaging the results in a spreadsheet as a way to smooth out that measurement variability. My suggestion is to incorporate an averaging mechanism in the plugin itself.

This would work as follows:

1. The UI would allow the user to choose the number of MBL runs to average over. Default would be '1' so a single run would be standard behaviour.

2. If the user selected say '5' runs, the MBL routine would run five times in succession with no user intervention. The UI could be updated to show Run 1/5....Run 2/5 etc.

3. A storage array is defined for the MBL measurement points. All cells in the array are zeroed at the start of each 'averaging' sequence.

4. For every run in the sequence, the results from each measurement point are added to the appropriate cell in the storage array to get a cumulative total for each measurement position.

5. Once all runs in the sequence have completed, the collected cumulative values are divided by the number of runs in the sequence to get the averaged values for each point.

6. The results are then displayed to the user in the normal way.

Subsequent mechanical adjustments would then be less 'hit or miss' as you try an home-in on the sweet spot.

Kind regards,

Tim Weston

scottrini, 2022-11-28:
Hey there,

I'll leave this up here in case anyone else wants to work on this, but it's not something I'd spend time on.  It's not a bad idea, though.

twes100, 2022-11-30:
Understood. I may even have a go at it myself if I get a bit if spare time :-)

Cheers,

Tim