scottrini/OctoPrint-PrusaLevelingGuide #50: Variance not calculated correct

vctrade, 2022-11-25:
When I manually calculate the variance (in this case, the difference between the smallest and largest number) I get a different variance than what is displayed. See attached the plugin shows a variance of 0,085 while the difference between the smallest and largest number is only 0,017

[image]

scottrini, 2022-11-25:
@vctrade Thanks for posting this!

I think you're definitely onto something here because I took a quick look and I think I immediately realized what's going on...
If you see [here](https://github.com/scottrini/OctoPrint-PrusaLevelingGuide/blob/master/octoprint_PrusaLevelingGuide/__init__.py#L105), it's calculating the variance using mesh_values, but these are the actual absolute values from the firmware.  It's converting these values to relative values and that's what you're seeing in the raw values table, the relative values.

When it's converting to relative values, the values are calculated relative to center.  Center is not usually zero, but since everything is relative to center, it is for these calculations.  So since it's calculating variance using the absolute values, but displaying the relative values, that could definitely account for that variation.

I can update this to calculate variance using the relative values.  I'll try to update it today, actually.

vctrade, 2022-11-25:
@scottrini thank you for the quick answer.

Does this also influence the actual adjusting of the bolts itself or is it only a mathematical thing? And why does this happen to me and not to all MK3S(+) users?

Note sure if it helps but these are the values from the G80/G81
[image]

scottrini, 2022-11-28:
No, this does not influence adjusting the screws.  It was happening to everyone, it's just a very slight skew and the amount it varies changes per user.  The variance is based on the **raw value** of the center point.  Not the value displayed in the plugin, that's actually set to 0 since you can't adjust the center screw, so that's the base and all other measurements are converted to be relative to that center point, to calculate how much to adjust it to **match** the center.

I've corrected this in [1.0.18](https://github.com/scottrini/OctoPrint-PrusaLevelingGuide/releases/tag/1.0.18).  In my local test on a printer, it dropped 0.49 in variance once the fix was applied.

So just to quickly reiterate, every other calculation in the plugin was being done with the calculated relative values, just that one line of python I linked was still referencing the original values from the firmware, so that was the issue.  So none of the other calculations were affected.

The reason this skew would vary per user is that the variance can be up to the raw value of the center measurement from the fw, which is normally zero'd out.  Everyone's center measurement will be slightly different, hence the variance in the skew.

closed 2022-11-28 by scottrini