scottrini/OctoPrint-PrusaLevelingGuide #51: New install on Raspbery pi 3 b+ will not connect to prusa mk3 with plugin installed

mattfox27, 2023-03-09:
Its wierd, brand new install with this plugin enabled it will not connect to prusa mk3, disable it and it connects fine

scottrini, 2023-03-10:
Yeah, that is pretty weird...I just setup a brand new 3b+ with octoprint and this plugin for someone I gave a mk3s+ to earlier this week...no issues.  So this would definitely have to be unique to your setup, since there are many current users of the plugin without the issue.  I'd take a look at the unique combination of plugins you have installed.

mattfox27, 2023-03-10:
I had to bump the serial connection timeouts up then it worked I guess it was just timing out to fast

scottrini, 2023-03-11:
well, I'm glad you figured it out at least!  Happy printing!

closed 2023-03-11 by scottrini