scottrini/OctoPrint-PrusaLevelingGuide #54: Not working after upgrading from mk3s+ to 3.5

dac8767, 2024-04-22:
This plugin was working perfectly with mk3s+, but after upgrading to prusa 3.5, it on longer works. The printer will take its measurements, but nothing is transmitted to octoprint. I gets stuck at “waiting for mesh level to complete”

jec0709, 2024-04-24:
Please update the firmware to work with MK3.5!!!!

scottrini, 2024-05-15:
The entire mod and process has been obsolete since 7x7 mesh bed leveling was added.  This was built prior to that.  There's no reason to update this for newer printers, but if someone wants to do it, I'm not going to stop them.

jec0709, 2024-05-16:
p{margin-top:0;margin-bottom:0}
    

    That's good newsI would really appreciate it if you could make itI'll be looking forward to itDue to problems with MK3.5, I'm now trying a different way to level my buds.--------- 원본 메일 ---------

  
    보낸사람: Scott Rini ***@***.***>
    받는사람: scottrini/OctoPrint-PrusaLevelingGuide ***@***.***>
      참조: jec0709 ***@***.***>,Comment ***@***.***>
    날짜: 24.05.16 05:32 GMT +0900
    
    제목: Re: [scottrini/OctoPrint-PrusaLevelingGuide] Not working after upgrading from mk3s+ to 3.5 (Issue #54)
    
    
  
  
    
The entire mod and process has been obsolete since 7x7 mesh bed leveling was added.  This was built prior to that.  There's no reason to update this for newer printers, but if someone wants to do it, I'm not going to stop them.

der-eine, 2024-05-21:
Hello Scottrini,

I read that you believe this approach no longer makes sense for newer printers. However, the leveling process of the MK3.5 is no different from that of the previous printers in the series. Only from the MK3.9 onwards is there a load cell and thus automatic leveling. Perhaps you could reconsider your opinion and take another look at it. Thank you.

Investegate, 2024-10-04:
Yea support for the MK3.5 would be very nice. Since the buddy board uses Marlin 2 with UBL.

scottrini, 2024-10-04:
I understand people are interested in support, but this will never be prioritized or built into this plugin unless someone else takes it on.  The only Prusa printers we have now are MK4S's and XL's, so I wouldn't get a 3.5 just to work on this.  I'll be updating the README.md to specify this only supports the MK3 and will not support the MK3.5.  If someone else forks the plugin and works on it, I wouldn't be opposed to merging a PR with the support if they submit one (and they'd obviously get credit for the contribution).

But for me, I wouldn't get a MK3.5 just to invest additional time into this, sorry.

closed 2024-10-04 by scottrini