scottrini/OctoPrint-PrusaLevelingGuide #56: Upgrade to 1.0.19 installs 1.0.18

thomergil, 2024-10-12:
I get a warning that I need to install 1.0.19. I press update. I see it download 1.0.19, but the last message says that version 1.0.18 is installed. After a restart, I, again, get a notice that I need to upgrade to 1.0.19.

thomergil, 2024-10-12:
[image]

[image]

[image]

[image]

scottrini, 2024-10-14:
Fixed in 1.0.20

closed 2024-10-14 by scottrini